RobotLocomotion/drake #6972: vtk drake_visualizer problems on OS X

TristanThrush, 2017-09-06:
Hi,

There seems to be a runtime problem with drake_visualizer and vtk on my Mac. I am using an up-to-date drake master at the time that this issue was created. 

Here is what I run:
```
bazel clean
brew update && brew upgrade && brew cleanup -a
./setup/mac/install_prereqs.sh
bazel run //tools:drake_visualizer
```
Here is the error:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
  File "/usr/local/lib/python2.7/site-packages/vtk/__init__.py", line 41, in <module>

  File "/usr/local/lib/python2.7/site-packages/vtk/vtkCommonCore.py", line 9, in <module>
ImportError: No module named vtkCommonCorePython
```
I then thought that the problem was possibly because vtk@8.0 was not linked, so I ran:
```
brew link --force vtk@8.0
```
Although it finds vtkCommonCorePython now, I get a new error when I try to run drake_visualizer:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_tristanthrush/4c74983c5d6e6b2655ec244fa3aa9720/execroot/drake/bazel-out/osx-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
  File "/usr/local/lib/python2.7/site-packages/vtk/__init__.py", line 115, in <module>
    from .vtkIOParallel import *
  File "/usr/local/lib/python2.7/site-packages/vtk/vtkIOParallel.py", line 9, in <module>
    from vtkIOParallelPython import *
ImportError: No module named vtkIOParallelPython
```

Apologies if I'm making a simple mistake - I'm still fairly new to drake.

soonho-tri, 2017-09-06:
FYI, this is the problem reported in https://github.com/RobotLocomotion/drake/pull/6903.

TristanThrush, 2017-09-06:
Okay; thanks. I'll just close this issue and follow #6903 until it is resolved.

closed 2017-09-06 by TristanThrush

soonho-tri, 2017-09-07:
@TristanThrush , as #6903 closed, could you try the following and let us know if it works this time?

```c+
brew update && brew upgrade && brew cleanup -a
./setup/mac/install_prereqs.sh
bazel run //tools:drake_visualizer
```

TristanThrush, 2017-09-08:
Yes; it works! Thanks.

54WC, 2019-05-05:
@TristanThrush , I follow your comment on 8 Sep 2017, but it can not cure the same issue at 2019. 
It is MAC & VKT@8.2. Could you kelp me ? 

Here is I run: 
brew update && brew upgrade && brew cleanup
./setup/Mac/install_prereqs.sh
bazel run //tools:drake_visualizer

Here is the result: 
INFO: Analysed target //tools:drake_visualizer (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //tools:drake_visualizer up-to-date:
  bazel-bin/tools/drake_visualizer
INFO: Elapsed time: 0.708s, Critical Path: 0.01s
INFO: 0 processes.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk.py", line 12, in <module>
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtkmodules/all.py", line 7, in <module>
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtkmodules/vtkCommonCore.py", line 9, in <module>
ImportError: No module named vtkCommonCorePython

xisun5, 2019-05-28:
I'm having the same issue with Mac OS and vtk 8.2. Have you figured it out yet? @54WC

54WC, 2019-05-28:
Not yet to find solution. ( After several nights trial and errors, I give it up )