RobotLocomotion/drake #6986: `bazel run valkyrie_balancing_controller_system` segfaults (with no useful errors) if gurobi is missing

RussTedrake, 2017-09-07:
I don't *think* this is a problem with the valkyrie example in particular.  I think this is a failure of the build system and/or mathematical program.  

Build system:
In short, a user can `bazel run` a test marked with the `gurobi_test_flags` turned on, even if they don't have gurobi installed.  No warnings or errors are printed.  We probably shouldn't even build the target in the first place?

QP controller or Mathematical Program:  
I haven't looked into it at all (just saw this happen on @ethanweber 's machine a few minutes ago), but we definitely shouldn't segfault even if that solver is missing and someone manages to run the executable.

jwnimmer-tri, 2017-09-07:
>  We probably shouldn't even build the target in the first place?

It is helpful to always compile the program, so that developers without Gurobi (the majority of Drake developers) receive compiler feedback if they change APIs or etc. and end up breaking the code.  Waiting for CI to test code that only depends on Gurobi _at runtime_ is frustrating.

Personally, I would be OK if we compiled it and then when you ran it it said "I'm missing the solver that can handle this" and exited with status 1.  I agree that a segfault is unwanted.

However, if you do want to remove the ability to run the program at all, we could imagine changing the build system to compile it into a non-runnable binary, or even just a library, when Gurobi was absent.

RussTedrake, 2017-09-07:
Got it.  OK, I agree that compiling it but printing a helpful solver message when it runs is better.  Do we handle that at the bazel level, or the c++ level?

jwnimmer-tri, 2017-09-07:
I think both?

The code should be written so that unexpected conditions turn into aborts, or exceptions, or printouts, or etc. as a general rule.  So someone should still figure out why things are segfaulting.  If its off in the weeds maybe we decide to ignore it, but if it's in some core library we should fix it.

But probably as a matter of convenience (so it can be applied broadly and quickly), we could have the build system replace the `main` function with one that immediately prints a diagnostic and fails.  (You could even have the replacement `main` offer a `--force-run` flag that overrode this.)  I think a linker `--wrap` flag would do the job.

siyuanfeng-tri, 2017-09-07:
I will add a throw in the qp controller to check for gurobi.

RussTedrake, 2017-09-08:
Could it go in mathematical program solver interface instead?  (To catch anyone that tries to launch a specific solver but gets the no_blah version instead)?