RobotLocomotion/drake #7003: Rework github.bzl to avoid git tgz checksums changing out from under us

jwnimmer-tri, 2017-09-12:
Our `master` was broken for a while, due to a GitHub OS upgrade.  We fixed it in #7002, but now bisecting is broken (modulo manual content-addressable-cache pre-seeding work-arounds), and we are still at risk of GitHub changing their implementation and disturbing the checksum again.

GitHub's summary of the intended semantics and hazards is posted at https://github.com/libgit2/libgit2/issues/4343#issuecomment-328631745.

We should update our "grab a certain commit from a github.com repository" implementation to avoid this pitfall.

jwnimmer-tri, 2017-09-12:
(1) One method would be to fall back to our own mirrors if the github.com checksum failed (with some kind of telemetry to remind us to bump the checksum and re-mirror the source archive).  Our own mirrors should stay intact even if github's download morphs.  (Possibly we'd have to add the checksum to our mirror filename, since we'll now have multiple concurrently-hosted artifacts with the same commit or tag.)

(2) Another related option would be to stop pulling from github.com entirely, and only ever refer to our self-hosted mirrored releases (or possibly first-party hosting, if the project offers permanent source archives, rather than on-demand generated tarballs).

(3) Another method would be to avoid `new_http_archive` and instead roll our own in skylark, where we verify the integrity of the download with our own checksum that inspects the unpacked contents, instead of the source tarball.

Other proposals?

jwnimmer-tri, 2017-09-12:
To +@stonier for delegation.

jamiesnape, 2017-09-12:
(1) would be easiest.

> (Possibly we'd have to add the checksum to our mirror filename, since we'll now have multiple concurrently-hosted artifacts with the same commit or tag.)

Probably not. We only need one. If it matches the GitHub SHA then great, and we only use it if the GitHub HTTP download fails, otherwise we know the SHA so we download and compare as before. If we notice the mirror getting a lot of hits, we can update the archive either manually, or my hope is that it would be eventually be automatic.

Putting the CloudFront URL first would not be such as bad idea. Downloads of newly changed references should fall through to the GitHub archive.

jwnimmer-tri, 2018-06-14:
@jamiesnape I think you have the best understanding of what (if anything) we should still improve, suggestions for next steps, and to help inform what the priority should be.

jwnimmer-tri, 2018-11-14:
Hearing nothing, seems ok to close.

closed 2018-11-14 by jwnimmer-tri