RobotLocomotion/drake #7015: Provide a single inclusion point for autodiff-related headers

soonho-tri, 2017-09-13:
Problem
---------

https://github.com/RobotLocomotion/drake/pull/7007 will introduce our own specialization for `Eigen::AutoDiffScalar` for `VectorXd` (a.k.a. `AutoDiffXd`). A consequence of this change is that we need to include this specialization before any code instantiates `Eigen::AutoDiffScalar<VectorXd>`.

Solution
--------

Provide a single inclusion point which includes the following in the right order:

```
<unsupported/Eigen/AutoDiff>
drake/common/autodiff_overloads.h
drake/common/autodiffxd.h
drake/common/eigen_autodiff_limits.h
drake/common/eigen_autodiff_types.h
drake/math/autodiff.h
drake/math/autodiff_gradient.h
```

We can use the same trick that we used in [drake/common/symbolic.h](https://github.com/RobotLocomotion/drake/blob/master/drake/common/symbolic.h) to detect if a user includes one of the above headers directly not via the representative header.

@jwnimmer-tri , I was thinking about naming this header as `drake/common/autodiff.h`. But we already have `math/autodiff.h`. Do you have a better suggestion? Also what do you think about moving the files in `math` directory to `common`?

jwnimmer-tri, 2017-09-13:
> I was thinking about naming this header as drake/common/autodiff.h. But we already have math/autodiff.h.

I see no problem with that.

> Also what do you think about moving the files in math directory to common?

I assume you mean the autodiff-related portion of math?  We can discuss that separately from the problem above.  It's a little tricky because math:autodiff and math:gradient and math:autodiff_gradient are all related, and possibly should travel together.  Probably should do a review of a call sites and make sure its all well-curated, unit tested, good docs, etc. moreso than file placement.

soonho-tri, 2017-09-13:
OK. Then I'll proceed with my proposal -- `drake/common/symbolic.h` will include the seven headers mentioned above.

soonho-tri, 2017-09-15:
/cc @rdeits and @EricCousineau-TRI

soonho-tri, 2018-06-14:
Close by #7199

closed 2018-06-14 by soonho-tri