RobotLocomotion/drake #7018: Document Eigen practices pertinent to Drake development

EricCousineau-TRI, 2017-09-14:
It'd be useful to have some Eigen practices / conventions documented, as it pertains to Drake.

Some examples:

* `Iosmetry3d` - use of `rotation()` vs. `linear()`, ensuring that it is compatible with glass-boxing
* `EigenPtr` - using it to preserve pointer syntax (when passing an output / a persistent reference)
* `resize` vs. `conservativeResize` - caveats when reshaping an array
* Alignment - may not be an issue that needs to be explicitly handled.
* Use of `auto` to not need to worry about mega-nested types - when you are certain that it will not adversely affect evaluation behavior / performance.

\cc @amcastro-tri @jwnimmer-tri

amcastro-tri, 2017-09-15:
I would also add a world of advice on whether to use `Eigen`'s `eulerAngles()` implementation or the methods we have in `math::`. I believe @hongkai-dai and @mitiguy worked on this issue in the past.

jwnimmer-tri, 2021-11-12:
If we find a re-occurring defect during code review where PRs that should use EigenPtr are not doing so, then we can update our style code and/or review checklist to recommend EigenPtr.  Therefore, I think this issue has no further work remaining.

closed 2021-11-12 by jwnimmer-tri