RobotLocomotion/drake #7020: Compilation errors on macOS High Sierra 10.13 Beta with Xcode 9 GM

jamiesnape, 2017-09-14:
```
ERROR: /Users/snape/Git/drake/drake/perception/dev/BUILD:25:1: Couldn't build file drake/perception/dev/_objs/feedforward_neural_network/drake/perception/dev/feedforward_neural_network.pic.o: C++ compilation of rule '//drake/perception/dev:feedforward_neural_network' failed (Exit 1).
In file included from drake/perception/dev/feedforward_neural_network.cc:1:
In file included from ./drake/perception/dev/feedforward_neural_network.h:10:
In file included from ./drake/perception/dev/neural_network.h:3:
./drake/systems/framework/leaf_system.h:217:64: error: call to unavailable member function 'value': 
      if (SystemSymbolicInspector* inspector_value = inspector.value().get()) {
                                                     ~~~~~~~~~~^~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/experimental/optional:547:17: note: candidate function has been explicitly made unavailable
    value_type& value()
                ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/experimental/optional:539:33: note: candidate function has been explicitly made unavailable
    constexpr value_type const& value() const
                                ^
In file included from drake/perception/dev/feedforward_neural_network.cc:1:
In file included from ./drake/perception/dev/feedforward_neural_network.h:10:
In file included from ./drake/perception/dev/neural_network.h:3:
./drake/systems/framework/leaf_system.h:233:61: error: call to unavailable member function 'value': 
            overridden_feedthrough ? overridden_feedthrough.value() :
                                     ~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/experimental/optional:539:33: note: candidate function has been explicitly made unavailable
    constexpr value_type const& value() const
                                ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/experimental/optional:547:17: note: candidate function not viable: 'this' argument has type 'const optional<bool>' (aka 'const std::experimental::fundamentals_v1::optional<bool>'), but method is not marked const
    value_type& value()
                ^
In file included from drake/perception/dev/feedforward_neural_network.cc:1:
In file included from ./drake/perception/dev/feedforward_neural_network.h:10:
In file included from ./drake/perception/dev/neural_network.h:3:
./drake/systems/framework/leaf_system.h:232:20: error: default initialization of an object of const type 'const bool'
        const bool direct_feedthrough =
                   ^
                                      = false
drake/perception/dev/feedforward_neural_network.cc:241:16: note: in instantiation of member function 'drake::systems::LeafSystem<double>::GetDirectFeedthroughs' requested here
template class FeedforwardNeuralNetwork<double>;
               ^
In file included from drake/perception/dev/feedforward_neural_network.cc:1:
In file included from ./drake/perception/dev/feedforward_neural_network.h:10:
In file included from ./drake/perception/dev/neural_network.h:3:
./drake/systems/framework/leaf_system.h:232:20: error: default initialization of an object of const type 'const bool'
        const bool direct_feedthrough =
                   ^
                                      = false
drake/perception/dev/feedforward_neural_network.cc:242:16: note: in instantiation of member function 'drake::systems::LeafSystem<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> > >::GetDirectFeedthroughs' requested here
template class FeedforwardNeuralNetwork<AutoDiffXd>;
               ^
4 errors generated.
ERROR: /Users/snape/Git/drake/drake/solvers/BUILD:661:1: Couldn't build file drake/solvers/_objs/complementary_problem_test/drake/solvers/test/complementary_problem_test.pic.o: C++ compilation of rule '//drake/solvers:complementary_problem_test' failed (Exit 1).
In file included from drake/solvers/test/complementary_problem_test.cc:1:
In file included from external/gtest/googletest/include/gtest/gtest.h:54:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/limits:110:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/type_traits:1301:38: error: incomplete type 'drake::solvers::detail::Constraint' used in type trait expression
    : public integral_constant<bool, __is_abstract(_Tp)> {};
                                     ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/type_traits:1374:39: note: in instantiation of template class 'std::__1::is_abstract<drake::solvers::detail::Constraint>' requested here
                                     !is_abstract<_T2>::value> {};
                                      ^
./drake/solvers/create_cost.h:50:36: note: in instantiation of template class 'std::__1::is_convertible<std::__1::shared_ptr<drake::solvers::QuadraticCost>, drake::solvers::detail::Constraint>' requested here
struct is_convertible_workaround : std::is_convertible<A, B> {};
                                   ^
./drake/solvers/create_cost.h:64:18: note: in instantiation of template class 'drake::solvers::detail::is_convertible_workaround<std::__1::shared_ptr<drake::solvers::QuadraticCost>, drake::solvers::detail::Constraint>' requested here
          bool, (is_convertible_workaround<F, C>::value) ||
                 ^
./drake/solvers/create_cost.h:97:33: note: in instantiation of template class 'drake::solvers::detail::is_binding_compatible<std::__1::shared_ptr<drake::solvers::QuadraticCost>, drake::solvers::detail::Constraint>' requested here
  static constexpr bool value = is_binding_compatible<F, Constraint>::value;
                                ^
./drake/solvers/mathematical_program.h:744:35: note: in instantiation of template class 'drake::solvers::detail::assert_if_is_constraint<std::__1::shared_ptr<drake::solvers::QuadraticCost> >' requested here
  typename std::enable_if<detail::assert_if_is_constraint<F>::value,
                                  ^
./drake/solvers/mathematical_program.h:850:12: note: while substituting deduced template arguments into function template 'AddCost' [with F = std::__1::shared_ptr<drake::solvers::QuadraticCost>, Vars = const Eigen::Ref<const Eigen::Matrix<drake::symbolic::Variable, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> > &]
    return AddCost(MakeL2NormCost(A, b), vars);
           ^
./drake/solvers/create_cost.h:85:7: note: forward declaration of 'drake::solvers::detail::Constraint'
class Constraint;
      ^
1 error generated.
```

soonho-tri, 2017-09-14:
I'm wondering if this is something we can re-reproduce in Linux with `clang-4` or `clang-5`.

jamiesnape, 2017-09-14:
Maybe with a libc++ of similar version.

soonho-tri, 2017-09-14:
I see.

jamiesnape, 2017-09-14:
```$ clang -v
Apple LLVM version 9.0.0 (clang-900.0.37)
Target: x86_64-apple-darwin17.0.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

```
$ otool -L libc++.dylib 
libc++.dylib:
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.0)
	/usr/lib/libc++abi.dylib (compatibility version 1.0.0, current version 400.7.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.0.0)
```

soonho-tri, 2017-09-14:
For the problem "`optional` doesn't have `value()`". I think we need to update https://github.com/RobotLocomotion/drake/blob/master/drake/thirdParty/boost/tcbrindle_cpp17_headers/stx/optional.hpp#L20-L24

soonho-tri, 2017-09-14:
I'll install Xcode 9 GM this evening and investigate the other problems.

soonho-tri, 2017-09-15:
Regarding `error: incomplete type 'drake::solvers::detail::Constraint' used in type trait expression`, I think a solution is to move https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/create_cost.h#L87-L103 to `drake/solvers/mathematical_program.h` as suggested in the TODO at line 83. @EricCousineau-TRI , could you confirm this or suggest a better solution?

soonho-tri, 2017-09-15:
FYI, here is a patch:

```diff
diff --git a/drake/solvers/create_cost.h b/drake/solvers/create_cost.h
index b7751fc78..cad030f42 100644
--- a/drake/solvers/create_cost.h
+++ b/drake/solvers/create_cost.h
@@ -79,29 +79,6 @@ struct is_cost_functor_candidate
                                        (!is_convertible_workaround<
                                            F, symbolic::Expression>::value)> {};

-// Forward declare this for the template used below.
-// TODO(eric.cousineau): Place the template below directly in mathematical
-// program.
-class Constraint;
-
-/**
- * Template condition to only catch when Constraints are inadvertently passed
- * as an argument. If the class is binding-compatible with a Constraint, then
- * this will provide a static assertion to enable easier debugging of which
- * type failed.
- * @tparam F The type to be tested.
- * @see http://stackoverflow.com/a/13366183/7829525
- */
-template <typename F>
-struct assert_if_is_constraint {
-  static constexpr bool value = is_binding_compatible<F, Constraint>::value;
-  // Use deferred evaluation
-  static_assert(
-      !value,
-      "You cannot pass a Constraint to create a Cost object from a function. "
-      "Please ensure you are passing a Cost.");
-};
-
 }  // namespace detail

 }  // namespace solvers
```

```diff
diff --git a/drake/solvers/mathematical_program.h b/drake/solvers/mathematical_program.h
index 746dcd118..3a671fdde 100644
--- a/drake/solvers/mathematical_program.h
+++ b/drake/solvers/mathematical_program.h
@@ -169,6 +169,26 @@ namespace solvers {
  * @}
  */

+namespace detail {
+/**
+ * Template condition to only catch when Constraints are inadvertently passed
+ * as an argument. If the class is binding-compatible with a Constraint, then
+ * this will provide a static assertion to enable easier debugging of which
+ * type failed.
+ * @tparam F The type to be tested.
+ * @see http://stackoverflow.com/a/13366183/7829525
+ */
+template <typename F>
+struct assert_if_is_constraint {
+  static constexpr bool value = is_binding_compatible<F, Constraint>::value;
+  // Use deferred evaluation
+  static_assert(
+      !value,
+      "You cannot pass a Constraint to create a Cost object from a function. "
+      "Please ensure you are passing a Cost.");
+};
+}  // namespace detail
+
 class MathematicalProgram;
```

soonho-tri, 2017-09-15:
@jamiesnape , after fixing the two issues (`optional` and `assert_if_is_constraint`), I can compile Drake with XCode 9 GM. I am wondering why I can't reproduce the `default initialization of an object of const type 'const bool'` problem though. Anyway, it seems that we're good to go.

PR #7028 (2017-09-15): Move detail::assert_if_is_constraint to MathematicalProgram
Fix #7020

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7028)
commit ce900bf0 (2017-09-15): Move detail::assert_if_is_constraint to MathematicalProgram

Fix #7020
closed 2017-09-15 by EricCousineau-TRI