RobotLocomotion/drake #7024: Add `directorPython` to `drake_visualizer` package installed binaries?

EricCousineau-TRI, 2017-09-14:
I'd like to have access to `directorPython` from the installed version of `drake-visualizer`.

@jamiesnape May I ask if this would be possible?

sammy-tri, 2017-10-05:
At some point in time (roughly August 2017), building `//tools:drake_visualizer` resulted in `directorPython` showing up under `drake_visualizer.runfiles`.  I don't know if it's original presence was an accident and/or if it was deliberately removed, but it would be nice to have it back in a supported fashion.

RussTedrake, 2017-10-24:
@jamiesnape -- were you able to weigh in on this one?  (apologies if I missed it)

jamiesnape, 2017-10-24:
Getting it into the binary install packages should be possible. It is just a case of ensuring it picks up the correct dependencies from within the drake install. It is on my TODO list once the CMake to Bazel transition is completed. Getting it functioning from the Bazel build is not something I have looked into yet.

sammy-tri, 2017-10-25:
FWIW, back when it was (for whatever reason) in `drake_visualizer.runfiles`, that version of `directorPython` ran correctly.

sammy-tri, 2017-10-25:
I've only just realized that no version of `directorPython` or `drake-visualizer` we've installed as a prebuilt binary has ever actually been compiled against drake (which would, yes, beg the question of "what version of drake were you expecting it to be built against?") when it was built...  which means it wouldn't be able to load robot models, which, if true, significantly lowers the priority of this issue for me.  It's probably that @EricCousineau-TRI still has a need for this though.

EricCousineau-TRI, 2017-10-25:
I'd like it namely to simplify the process of viewing images over LCM; however, it is not a strong need.

>  has ever actually been compiled against drake

I believe that only eliminates the usage of the `dd*` classes that interface with Drake (and the utility functions based on it thereafter).
Are you able to get by with what `pydrake` offers, since it's available via install (and I believe is a dependency of `drake_visualizer`)?

sammy-tri, 2017-10-25:
Getting `TeleopPanel` (among other things) to work without `ddDrakeModel` would be a pretty major refactoring of director.  I suspect all the functionality could be implemented using `pydrake.rbtree` (the underlying IK code is using pydrake's IK already), but I'm not sure I'm inclined to do it...

jwnimmer-tri, 2018-06-09:
Relates #7321.  Relates #6229.