RobotLocomotion/drake #7037: Generated BasicVector subclasses should automatically generate SystemConstraints

jwnimmer-tri, 2017-09-18:
Generated BasicVector subclasses such as [PendulumParams](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/pendulum/gen/pendulum_params.h) should automatically generate [SystemConstraint](http://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system_constraint.html) instances when they are declared as State, Params, etc. of a System.

jwnimmer-tri, 2017-09-18:
WIP will be at https://github.com/jwnimmer-tri/drake/tree/vectorgen-constraint

jwnimmer-tri, 2017-09-19:
Since we want to have this on by default, it looks like we'll need constraint expression as a new virtual method on `VectorBase`.  I have a draft of that in the WIP branch.

I was hoping to use compile-time interrogation to see if the model vector had constraints, but it seems too likely to silently not work, and will be hard to, e.g., delegate declarations with constraints and preserve compile-time interrogation.  Instead, at runtime we'll ask if there are any constraints when the model vector is declared into a LeafSystem, and if so then we'll add the SystemConstraint.  I have a draft of that in the WIP branch.

sherm1, 2017-09-19:
FWIW this seems like a good idea to me and the WIP implementation looks reasonable.

jwnimmer-tri, 2017-09-20:
Per discussion in #7057, we should also declare constraints when the Output is typed.  I will re-open this to track that addition.

PR #7057 (2017-09-20): Teach LeafSystem about named_vector declared inequality constraints
Closes #7037.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7057)
PR #7067 (2017-09-25): Support named_vector inequality constraints on output ports
Closes #7037 again.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7067)
closed 2017-09-25 by soonho-tri