RobotLocomotion/drake #7045: Where to put hardware related lcmtypes?

siyuanfeng-tri, 2017-09-18:
We are experiencing some dependency pain with hardware related lcm message types between drake master and individual drivers (iiwa / schunk gripper / optitrack). We currently have three ways of dealing with it: 
For iiwa: the messages live inside drake master, and robot driver is directly dependent on drake master.. This feels really weird because you should be able to talk to the robot without building the entire drake. 
For schunk: the messages are duplicated in both drake master and driver. This is just bad. 
For optitrack: the messages live in the driver, and drake master depends on the driver. This is also weird because of potential external third party dependencies that come with the driver (.h / .so). We can get away with optitrack because it is essentially translating network packets from some windows computer that runs the third party software. So we avoid the dependency issue completely.  

I briefly talked with @sammy-tri, and I am leaning towards having a separate lcm type only repo that's just for these hardware related (or peripherals) message types. Something similar to https://github.com/RobotLocomotion/lcmtypes, but more organized and targeted. This is becoming an issue since we are likely to have more arms, grippers and sensors from different vendors. 

cc @sammy-tri @naveenoid @manuelli @RussTedrake @jamiesnape @jwnimmer-tri Do you guys see any potential issues? or other suggestions?

jwnimmer-tri, 2017-09-18:
> For iiwa: the messages live inside drake master, and robot driver is directly dependent on drake master.. This feels really weird because you should be able to talk to the robot without building the entire drake.

Without intending to argue for the iiwa case as the correct solution, I'd like to point out that "building the entire drake" is a strawman downside.  For one, you can build a subset of Drake if you only need a subset.  But more importantly, don't confuse where the `*.lcm` files live under source control from which build system creates the `*.{h,cpp}` files.  Some Driver could easily download a copy of Drake but never use Drake's build system, merely reaching inside the source archive to find the `*.lcm` files.  Similarly if you decide to put the files somewhere else, Drake could download the `*.lcm` files and compile the libraries it needs on its own.

manuelli, 2017-09-18:
I agree with @siyuanfeng-tri. I recommend a separate repo for hardware specific LCM types. In general I don't think that hardware specific lcmtypes should necessarily be committed to drake/master. My view is that robots will always need robot/hardware specific drivers/code/message types. This shouldn't live in a core library, but rather some lightweight external.

sammy-tri, 2017-09-18:
>  I'd like to point out that "building the entire drake" is a strawman downside.

You're right, it's merely that all of drake needs to be downloaded, in which case we could write our own partial build or do a partial build of drake (The example in https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_bazel_external which I was not previously aware of is interesting.  Perhaps I'll spike test something like this on the schunk driver and see if it works)

>  Similarly if you decide to put the files somewhere else, Drake could download the *.lcm files and compile the libraries it needs on its own.

Is this not what would happen if drake depended (via WORKSPACE) on a driver external, but only referenced the lcm types in BUILD files?

jwnimmer-tri, 2017-09-18:
> You're right, it's merely that all of drake needs to be downloaded, ...

Yeah, it's a slightly big download, but you'd only need to do it once (ish).  That the download is ~50MB is a fair reason to disprefer this option, though.  But if you need some robot models anyway, then it might tip the other way.

> Is this not what would happen if drake depended (via WORKSPACE) on a driver external, but only referenced the lcm types in BUILD files?

Yes, more or less.  The distinction is whether or not Drake's `tools/myhardwarelcmtypes.BUILD` is just a `filegroup()` that exports the files, or whether it also contains `lcm_cc_library()` calls to compile the code.  The compilation doesn't strictly have to happen within the external; it could be in `drake/manipulation/lcmtypes/BUILD` for example.

jwnimmer-tri, 2017-09-18:
But stepping back, my personal vote would be for https://github.com/RobotLocomotion/lcmtypes as the source tree to put the hardware driver types inside.  It's already a bag of lcmtypes, its already lightweight, its already hosted in RobotLocomotion, and its already public.  You can add a new folder in there if you like.  What else would you need?

siyuanfeng-tri, 2017-09-18:
@jwnimmer-tri I want to start fresh though because rl/lcmtypes were mostly for plan / control / visualization, and has some legacy stuff there as well. The new driver types doesn't really have anything to do with rl types.

siyuanfeng-tri, 2017-09-18:
@jwnimmer-tri

sammy-tri, 2017-09-18:
I tried a spike test of depending on drake in the schunk driver to get the lcm types.  It's a little brittle, because there are various files which have to be maintained in both `WORKSPACE`s...  e.g. drake's `tools/lcm.BUILD` and my `tools/lcm.BUILD` had gotten out of sync and drake was using newer names for things.  I suppose that's not the end of the world.   (also the 50MB drake download wasn't so bad, and it does only build enough to get the lcm types out of drake).  Not necessarily arguing for this approach, but it can work.

jwnimmer-tri, 2019-03-04:
@sammy-tri Can I nominate you to drive this issue home to completion?  Either resolving what is still painful with PRs or documentation, or deciding that it's not painful and closing the issue?

sammy-tri, 2019-03-04:
Ok, I'll come up with something or close.

sammy-tri, 2019-03-22:
I think I'm gonna go with RobotLocomotion/lcmtypes for robots whose lcmtypes currently live in drake but have external drivers depend on drake solely for their lcmtypes (and these drivers often are repositories under RobotLocomotion anyway) (so probably iiwa, schunk, jaco)  unless there's strenuous objection.  I may not bother with optitrack.

sammy-tri, 2019-04-01:
Hmmm, I forgot that the bazel builds for the iiwa/schunk drivers also use drake to provide their workspace in addition to the lcmtypes.  I'll think about how I feel about either pulling in all of drake just for that, or duplicating the workspace stuff (which would not be ideal, I used to keep it in sync manually and it was tedious).

jwnimmer-tri, 2020-04-28:
@sammy-tri Any new thoughts on this, as you're in the middle of a new robot / driver?

sammy-tri, 2020-04-28:
Sadly, no.  The new driver is still pulling in drake to get it's bazel workspace, so having the lcmtypes somewhere else isn't all that compelling for that use.

From slack, a couple of weeks ago:

> sammy-tri  16:07
> https://github.com/RobotLocomotion/drake/issues/7045  is still open for the question of where to put hardware related lcm types.  Given that I'm probably about to add a type for the Panda's status, what's your though on just giving up and putting it in drake vs actually trying to make progress on moving stuff out (given https://github.com/RobotLocomotion/drake/issues/7045#issuecomment-478731470)
> siyuan.feng  22:18
> I don't have a strong opinion either way now, stick it with drake is probably easier.
> sammy-tri  08:17
> yeah.   thanks.

jwnimmer-tri, 2020-04-28:
Seems like we can close the issue, in that case?  We all think `drake/lcmtypes` is good enough.

closed 2020-04-28 by sammy-tri