RobotLocomotion/drake #7050: CI coverage of pre-built drake visualizer

jwnimmer-tri, 2017-09-19:
Drake's CI should offer coverage of pre-built drake visualizer -- for example, that is launches without faults, that it runs for a few seconds without crashing, that it is able to locate data resources.  It is not unusual for users to experience problems running this software, so we should have CI coverage to point to that demonstrates that it works (and helps us keep regressions off of master).

Relates to #5621 as the feature to be tested in CI.

Relates to #6834 as an example of a scenario that should be tested in CI.

jwnimmer-tri, 2018-11-14:
Probably would be good enough to run it without opening a window, and then just immediately quit -- so that we test import / load paths.

jamiesnape, 2018-11-14:
It would if that we currently possible. We could however test the imports used by the launcher one level below.

jwnimmer-tri, 2018-11-14:
That SGTM to me too.

EricCousineau-TRI, 2019-01-10:
Just encountered an issue because this wasn't tested: We forgot to install the Python2.7 version of LCM, so now the installed `drake_visualizer` for `--config=python3` is broken.
However, I confirmed that `--help` doesn't catch this :(

We can just test imports in `drake_visualizer.py` wrappers as mentioned in #8324, but it means we won't catch install mistakes like this.

EricCousineau-TRI, 2019-01-10:
It would catch it for Bazel testing, but it wouldn't catch it at install, because we install the `drake-visualizer` binary, not a wrapper script.

patmarion, 2019-01-11:
hello!

is drake supporting python3 these days? if you're interested, there is python3 support for director.  it's not been merged to `master`, because there are still users of `master` stuck on python2, but you can get it here if needed- https://github.com/patmarion/director/tree/update-to-python3-ver5

jamiesnape, 2019-01-11:
Not on all platforms, so `drake-visualizer` is still Python 2.

EricCousineau-TRI, 2019-05-16:
Recent pain point: https://github.com/RobotLocomotion/drake/pull/11465#pullrequestreview-237565752

EricCousineau-TRI, 2020-04-23:
I want this to be narrowed down to just be `--help`.

jamiesnape, 2020-04-23:
`—help` still pops up a window that disappears straightaway, I think. Also if you look up in the comments, help was not sufficient (you said that, even).

EricCousineau-TRI, 2020-04-23:
I vaguely remembered something like that, but did not see the window when I ran it. Can I ask how you confirmed there's a window?

(If I run it on my machine, I see no window crop up, at least as of 629db0fcc on Ubuntu 18.04)

EricCousineau-TRI, 2020-04-23:
Looking at the code, I don't believe it currently starts a window (not sure what I encountered a year ago).

Tracing it out, my assumption is that the process exits before the app starts (which is responsible for instantiating the physical window on your WM?):
Parsing effectively happens through this (loosely identified) chain:
https://github.com/RobotLocomotion/director/blob/4c3e570a6797ff840c74067c742455daaa113d93/src/python/director/drcargs.py#L30-L33
https://github.com/RobotLocomotion/director/blob/4c3e570a6797ff840c74067c742455daaa113d93/src/python/director/drakevisualizerapp.py#L32-L36

So I think we're good?

EricCousineau-TRI, 2020-04-23:
Also confirmed that Drake Viz will fail if it can't access a display, even on these simplified codepaths, so maybe `--help` is more-than-decent enough test?
```
env DISPLAY=: ./bazel-bin/tools/drake_visualizer --help
```

jamiesnape, 2020-04-24:
I think it is useful, so I have no problem with us having that test initially. I would need to go through the issues to remind myself what the main point of failure was. OS and VTK upgrades are high risk for `drake-visualizer`, so now is the time to add something.

PR #13127 (2020-04-29): drake_visualizer: Install builtin scripts
Put under test, both in Bazel and install

Resolves #10486 
Resolves #7050.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13127)
closed 2020-04-29 by EricCousineau-TRI