RobotLocomotion/drake #7058: LeafSystem::DeclareDiscreteState should offer a model_value overload

jwnimmer-tri, 2017-09-20:
`LeafSystem::DeclareDiscreteState` currently only offers an `(int)` overload, that dimensions a `BasicVector` to hold the discrete state in the Context.  We should offer a `model_value` overload (similarly to `DeclareContinuousState`, `DeclareNumericParameter`, etc.) where a `BasicVector` subtype can be used.  That new overload should likewise forward along any inequality constraints, like the other `model_value`-based declaration APIs.

edrumwri, 2017-09-20:
Jeremy, @sherm1 thought your issue would be a good place to note a somewhat related issue that I encountered this morning:

I have two (diagram) systems identical but for the fact that one uses discrete state and one uses continuous state. I want to simulate both systems and compare the result, which I attempt to do via the following:

`  auto& continuous_state = continuous_sim.get_context().get_continuous_state_vector();`
`  auto& discrete_state = discrete_sim.get_context().get_discrete_state_vector();`
`  ASSERT_EQ(continuous_state.size(), discrete_state.size());`
`  for (int i = 0; i < discrete_state.size(); ++i) {  EXPECT_NEAR(discrete_state[i], continuous_state[i], 0); }`

I get an assertion failure on the `get_discrete_state_vector()` line:
`
abort: Failure at ./drake/systems/framework/discrete_values.h:84 in get_vector(): condition 'num_groups() == 1' failed.
`
because discrete variables are organized into groups (causing `get_discrete_state_vector()` to fail).

jwnimmer-tri, 2017-09-20:
I think that's not related.  How many groups were there -- zero? three?  The sugar proposed above would just create one group (which is the default for `LeafSystem::DeclareDiscreteState` already).

sherm1, 2017-09-20:
Sorry I was thinking of this as "discrete variables need some repair". There are other problems with this class that I've got fixed in my caching branch but will have to extract. I'd be happy to include both the model vector and a Supervector gathering all the discrete vars with that, or these could be treated independently.

jwnimmer-tri, 2017-09-20:
@sherm1 I think the earlier that the various improvements from your branch can be PR'd incrementally, the better off we will all be.  So it probably makes sense to rollup this request into the repairs you mention, at least unless someone clamors for it earlier.  (As I understand it now, this is "we should tick this checkbox" feature request, but there's no acute example where we need it right away.)

sherm1, 2017-09-20:
Yes, that's my plan this time @jwnimmer-tri -- lots of small PRs. I'll make this one of them.

RussTedrake, 2017-10-01:
One more issue: even the existing interface (taking just the vector size) is misleading. I would have thought that if I called DeclareDiscreteState twice, I would have gotten two distinct discrete state groups. In fact, I would just get one, with the size of the last call do DeclareDiscreteState. Yikes!

sherm1, 2018-11-28:
Lack of model value made our difference equation example uglier in #9766. I'll bump up the priority on this.

PR #10136 (2018-12-05): Permit multiple discrete state groups and model vectors for each
Modifies `LeafSystem::DeclareDiscreteState()` to create a new discrete state group each time it is called, and permits model values to be provided for them. The DiscreteStateIndex of the group is returned.

Changes here should be backwards compatible unless someone was depending on multiple DeclareDiscreteState() calls to be ignored except for the last one (now you'll get multiple groups instead).

Reviewer notes:
- Required adding a method to DiscreteValues to allow groups to be added incrementally.
- DeclareDiscreteState() beha [... cut]
closed 2018-12-05 by sherm1