RobotLocomotion/drake #7077: mosek license errors fail silently

RussTedrake, 2017-09-22:
Despite the code at:
https://github.com/RobotLocomotion/drake/blob/709171b00b92e79557920329346f5500718a2caf/drake/solvers/mosek_solver.cc#L588-L596._
when my mosek license expired, i was getting numerical errors in my multiple_shooting_test instead of an announcement of the license failing.  I verified that the above code DID get called, but that the return code was still `MSK_RES_OK`.  

To reproduce, simply move `~/mosek/mosek.lic` to `~/mosek/mosek.lic.bk` and run `multiple_shooting_test`.

jamiesnape, 2017-09-22:
I think you need something like `MSK_initenv` for MOSEK to actually check the license.

sammy-tri, 2017-09-22:
I did some digging here, and I believe Jamie is correct that `MSK_makeenv` does not check the license, so the throw above is misleading at best.

I think we could add `MSK_initenv` to check the license when the environment is first created.  As it is now, we don't get an error until we call `MSK_optimizetrm` later in `Solve`.  However, in my testing, this error is detected and `Solve` correctly returns an error.

Is part of the confusion the fact that `multiple_shooting_test` does `EXPECT_EQ(prog.Solve(), solvers::SolutionResult::kSolutionFound);` instead of `ASSERT_EQ`, so it's still testing the numerical results (and displaying errors) even though we've already failed?

Other question...  If a license is not available, do we want to throw?  Or is returning an error from `Solve` (possibly with a log message indicating the license failure) sufficient?

EricCousineau-TRI, 2017-09-22:
I think it'd be most appropriate to make the throw happen when we try to acquire the license.
In most cases (e.g., in the `MathematicalProgram::Solve`), each solver instance is instantiated, but for `MosekSolver`, license acquisition is not attempted until `Solve`.

Additionally, if the user acquires the license at a larger scope (using `MosekSolver::AcquireLicense()`), then I think it'd be most informative to error out then and there, to keep expectations simple.

sammy-tri, 2017-09-22:
I guess if we want to make `MosekSolver::License` not a total lie about what it does, we should `MSK_initenv` there and throw.  I'll try a spike test of this in a bit.

jamiesnape, 2017-09-22:
If we set `MOSEKLM_LICENSE_FILE` to some junk location, can we simulate not having a valid license, and allow us to have a unit test for this code?

sammy-tri, 2017-09-22:
Testing indicates that `MSK_initenv` can still return `MSK_RES_OK` even with the license file missing, despite http://docs.mosek.com/7.0/capi/MSK_initenv_.html indicating that the license server will be contacted.  At this point, I hesitate to assume there's any guarantee regarding which `MSK_*` calls actually involve license acquisition.

My current plan is to update the documentation for `MosekSolver::License` (and possibly rename it) to explain that it's actually just RAII on the environment for thread safety and doesn't (necessarily!) perform a license check.  I'll also update `multiple_shooting_test` to use `ASSERT_EQ` while I'm in there.

jamiesnape, 2017-09-22:
Does this work to specification? http://docs.mosek.com/7.0/capi/MSK_checkoutlicense_.html

sammy-tri, 2017-09-22:
Yes!  Nice find, I'll go that route.

PR #7079 (2017-09-22): Make MosekSolver::License acquire the license
Fixes #7077

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7079)
closed 2017-09-22 by sherm1