RobotLocomotion/drake #7086: Use vector gen's `GetCoordinateNames()` method in SystemSymbolicInspector

RussTedrake, 2017-09-24:
Unfortunately, this method seems to be *never* used (apart from one coverage test) and appears hard to use in a general systems' framework, because it is only defined for the generated types.  (`BasicVector` has no understanding of CoordinateNames, so I would have to downcast to the specific known type to use it).  

Proposal:  I'll add a method:  `virtual optional<const vector<string>&> GetCoordinateNames() const` to `BasicVector`, with the default implementation returning `nullopt`, then modify the vector gen method to override this method.  Note that this implementation would mean that `GetCoordinateNames()` is no longer a `static` method, which was good, but I believe is incompatible with my proposal.  

@jwnimmer-tri , @sherm1 -- seem reasonable?

jwnimmer-tri, 2017-09-24:
Seems mostly good, though having the motivating example spelled out might help clarify the API choices.

For the codegen, I would probably leave, e.g., `DrivingCommandIndices::GetCoordinateNames` alone (still a static method), but then `DrivingCommand::GetCoordinateNames` could be a non-static override method of `VectorBase`, whose body simply delegates to the `-Indices` implementation for the generated vectors.

Also worth considering for the `VectorBase` API is if the result is allowed to be sparse -- you could imagine an e.g. Multiplexer (or Supervector) that only has coordinate names for some items. Possibly the `VectorBase` API should allow expressing that?  An `optional<vector<string>>` doesn't really admit naming sparsity, unless we make up dummy names for the unknown items.  (In which case, possibly we should lose the `optional<>` altogether, and have `[0]` and `[1]` and etc. as default names in `VectorBase` already.)

RussTedrake, 2017-09-24:
Yes, default names like "v1" "v2" would be the other option. The reason i did not recommend that at first is that i think systems/algorithms consuming he vectors are in a better position to provide defaults than the vector itself (eg Pendulum's state might be x in one case but u in another another; but i would always prefer theta.  But i could be convinced.

I would like to go up to vectorbase... as the value only increases as we get to diagrams.  Presumably the element names could get a prefix from their system to make them unique (though i was not going to enforce this).

RussTedrake, 2023-02-27:
This is no longer our intended workflow; I understand it to be replaced by #12566.

closed 2023-02-27 by RussTedrake