RobotLocomotion/drake #7117: Profile and speed up LCM image publishing

EricCousineau-TRI, 2017-09-27:
While producing a few demos, we found that now the bottlenecks in the RGB-D camera simulation pipeline is most likely publishing LCM images. This is most likely due to compression, but we'll look into this and confirm.

\cc @kunimatsu-tri

EricCousineau-TRI, 2017-09-28:
Seems like it was really just an issue due to #5629 - I was testing in `monolithic_pick_and_place_demo`, which did not have `simulator.set_publish_every_time_step(false)`.

Will check and see if there is any other low-hanging fruit.

hongkai-dai, 2019-03-25:
@EricCousineau-TRI could you update the status of this issue? Thanks!

EricCousineau-TRI, 2019-03-25:
I think this one is more-or-less resolved (in that it hasn't been identified as a major pain point).

Can either re-open or open a new issue if it does become an issue.

closed 2019-03-25 by EricCousineau-TRI