RobotLocomotion/drake #7122: Frames mistakenly registered with the same "name" do not show up in visualizer.

amcastro-tri, 2017-09-27:
A `GeometrySystem` problem. In my case I register a single frame per single geometry. I do frame registration by calling:
```c++
FrameId link_frame_id = geometry_system->RegisterFrame(
          source_id_, GeometryFrame(
              "MistakelyAlwaysTheSameName", Isometry3<double>::Identity()));
```
Then, after calling the `geometry::DispatchLoadMessage(*geometry_system);`, I run the simulation. 
With that setup, only my first two geometries are shown and the rest are invisible.

If I now make the `GeometryName` different for each frame, life is good. Should that registration throw for frames with the same name?

amcastro-tri, 2017-09-28:
To be more specific with an example soon-to-be in master, `SolarSystem`. If I change the name "Luna" to "Earth", "Luna" is no longer rendered (I have no idea if it actually did make it to my geometry model though). I think the safest thing to do would be to require (throw if not) unique names.

SeanCurtis-TRI, 2017-09-28:
Thanks -- I'll get this fixed.

PR #17243 (2022-05-25): [geometry] Reject duplicate-named frames for same source
Closes: #7122

Track the set of frame names for each source, and use that to reject
duplicates. Add a unit test for the new message, and adjust some other tests
that needlessly used duplicate names.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17243)
commit df41d02e (2022-05-25): [geometry] Reject duplicate-named frames for same source (#17243)

* [geometry] Reject duplicate-named frames for same source

Closes: #7122

Track the set of frame names for each source, and use that to reject
duplicates. Add a unit test for the new message, and adjust some other tests
that needlessly used duplicate names.
closed 2022-05-25 by rpoyner-tri