RobotLocomotion/drake #7189: `region_of_attraction_test` on Sierra

m-chaturvedi, 2017-10-04:
[`mac-sierra-clang-bazel-nightly-everything-release`](https://drake-jenkins.csail.mit.edu/view/Mac%20Sierra/job/mac-sierra-clang-bazel-nightly-everything-release/68/consoleText) fails to build due to `region_of_attraction_test`, it is a MOSEK related test. @EricCousineau-TRI reported that:

The test 'passes' when run with `bazel test` as well as with `bazel run`  in `dbg` mode.  The test passes with `bazel run` in `opt` (default) mode but fails with `bazel test` in `opt` mode. 

This is the debug build, which is successful, [`mac-sierra-clang-bazel-nightly-everything-debug`](https://drake-jenkins.csail.mit.edu/view/Mac%20Sierra/job/mac-sierra-clang-bazel-nightly-everything-debug/68/)

soonho-tri, 2017-10-04:
I think I've fixed the problem in #7188 .

soonho-tri, 2017-10-04:
Now, it's merged. I forgot to run `mac-sierra-clang-bazel-nightly-everything-release`. Could you test it by triggering the build?

m-chaturvedi, 2017-10-04:
Scheduled:
https://drake-jenkins.csail.mit.edu/view/Mac%20Sierra/job/mac-sierra-clang-bazel-experimental-everything/

soonho-tri, 2017-10-04:
It’s blue.

closed 2017-10-04 by soonho-tri