RobotLocomotion/drake #7194: Extended interface for `maliput::multilane::Builder`

maddog-tri, 2017-10-05:
The `Builder` API (of `monolane` originally, and now `multilane`) exists to solve some end-to-end connection constraints when assembling a `RoadGeometry`.  `multilane` introduces more degrees-of-freedom than `monolane`, because of multiple lanes and variable offset of lane centers from the segment's "reference line".  (In `monolane` the centerline of the single lane is exactly equal to the reference-line.)  Elevation is expressed as a cubic polynomial along that reference line, and superelevation is a rotation around that reference line.  I see two basic flavors of constraints for hooking up segments end-to-end:  one based on lane-to-lane correspondence, the other based on reference-line correspondence.

The lane-to-lane version is probably would get the most general use, and reads something like this:

> Create a new `Connection`...
>  * of N lanes
>  * given lane `n_0` starts at a specific `Endpoint`, `E_0`
>  * given lane `n_1` ends with its out-of-plane pose specified by an `EndpointZ`,  `Z_1`
>  * given reference-line at offset `r_ref` relative to lane `n_ref`
>  * given a reference-line shape specified by a displacement which is either:
>    * linear, specified by a length `L`, or
>    * arc, specified (via an `ArcOffset`) by an arc-length `phi` and a radius `R` of the center from the reference-line

The initial `Endpoint` (`E_0`) could come from new per-lane accessors on `Connection`:
  * `Endpoint Connection::lane_start(int lane_index)`
  * `Endpoint Connection::lane_end(int lane_index)`

That would ensure correspondence of lanes from one connection to another.

The reference-line flavor of interface is similar, but focuses on directly specifying geometry of the reference-line:

> Create a new `Connection`...
>  * of N lanes
>  * given the reference-line starts at a specific `Endpoint`, `E_0`
>  * given the reference-line ends with its out-of-plane pose specified by and `EndpointZ`, `Z_1`
>  * given lane `n` has an offset `r0` from the reference-line
>  * given a reference-line shape specified by a displacement which is either:
>    * linear, specified by a length `L`, or
>    * arc, specified (via an `ArcOffset`) by an arc-length `phi` and a radius `R` of the center from the reference-line

If `E_0` came from `start()` or `end()` of an existing `Connection`, then one would want `r0` relative to that `Connection`, too, which involves another new accessor:
  * `double Connection::lane_offset(int lane_index)` --- return the offset of a lane from the reference-line

`monolane` is basically a degenerate form of `multilane`... and both these flavors collapse into the `monolane` version when `N = 1`, `n_0 = n_1 = n_ref = n = 1`, `r_ref = r0 = 0`.

basicNew, 2017-10-05:
cc @agalbachicar @hidmic

agalbachicar, 2017-10-06:
To provide a complete list of things to change and have some sort of road map of changes that `Builder` interface should suffer I'm writing this down:

**Lane-to-lane version:**

- Provide appropriate constructors to populate information on the `Connection` accordingly. Start and end `EndpointZ` are necessary in both cases (lane-to-lane and reference-line versions) but the curve to which they serve may not be same (lane centerline or reference-line).
- `Connection` needs to provide `Endpoint Connection::lane_start(int lane_index)` and `Endpoint Connection::lane_start(int lane_index)`.
  - It will require Segment's `RoadCurve` object to be previously constructed since it holds all the information to compute those `Endpoints`.
  - It will require the `Connection` to be the owner (or at least know about) of the `lane_width` (taking into account that lane spacing is equal to lane width) so as to translate `n_ref` to any Lane centerline.

**Reference-line version:**

- Provide `double Connection::lane_offset(int lane_index)` method. I realized that this requirement is met when the Connection knows about  the `lane_width` or is the owner of it. A pretty similar requirement as `lane_start` or `lane_end` methods. In addition, the `RoadCurve` should be previously built and known by the `Connection`.

**RoadCurve:**

- `RoadCurve` has `W_prime_of_prh()` method but it lacks of `W_of_prh()`. I think that by moving the current `multilane::Lane::DoToLanePosition()` implementation into `RoadCurve` should be enough (leaving only the _s --> p_ conversion) and would make `RoadCurve` more concise.

I wonder if a similar change should be done on the `multilane::Lane::DoGetOrientation()` method.

maddog-tri, 2017-10-10:
> **Lane-to-lane version:**
>
> - Provide appropriate constructors to populate information on the `Connection` accordingly. Start and end `EndpointZ` are necessary in both cases (lane-to-lane and [...]

"Start `Endpoint` and end `EndpointZ`..."
> - `Connection` needs to provide `Endpoint Connection::lane_start(int lane_index)` and `Endpoint Connection::lane_start(int lane_index)`.

"and  `Endpoint Connection::lane_end(int lane_index)`."
>   - It will require Segment's `RoadCurve` object to be previously constructed since it holds all the information to compute those `Endpoints`.

True... `Builder` currently defers most of the actual building of anything until `Build()` is called, and `Connection` only needs to figure out an `EndpointXy` when constructed.  Now, more math will be involved in supporting subsequent connections, and each `Connection` will need to have a `RoadCurve` available.  I can see most of the code in `Builder::BuildConnection()` moving into `Builder::Connect()` instead.  `Connection` itself will probably get more simple.
>   - It will require the `Connection` to be the owner (or at least know about) of the `lane_width` (taking into account that lane spacing is equal to lane width) so as to translate `n_ref` to any Lane centerline.

The `Builder` should still own the single `lane_width` value for the whole network (but since Builder builds its own `Connection`s via `Builder::Connect()`, it could give each Connection a copy, or each Connection could get a pointer back to the Builder that created/owns it).
> **Reference-line version:**
>
> - Provide `double Connection::lane_offset(int lane_index)` method. I realized that this requirement is met when the Connection knows about  the `lane_width` or is the [...]

Did you mean `multilane::Lane::DoToGeoPosition()`?  (That's what is basically doing "W_of_prh()".)
> 
> 
> I wonder if a similar change should be done on the `multilane::Lane::DoGetOrientation()` method.

Yeah, makes sense.  That leaves `multilane::Lane` really just doing the translations from lane-frame to the road-curve/segment coordinates.

agalbachicar, 2017-10-11:
Thanks for the comments. 

Yes, I did. I changed the original description so as to have a unified and correct to-do list.

agalbachicar, 2017-11-06:
Pending API changes after #7343 is merged:

**Lane-to-Lane version -- Builder API**

- Provide a method that creates an N-lane connection whose lane `X` starts at start / end `Endpoint` of connection `c` and lane `Y` ends with `EndpointZ`.

agalbachicar, 2017-12-07:
@maddog-tri  I guess it makes sense to continue with the [discussion](https://reviewable.io/reviews/robotlocomotion/drake/7471#-L-G5FxctxRnfSTZFX3Y) on this thread.

Below there are some translations from code to paper of the involved fomulas:

[image]

[image]

[image]

The problem is much more complex when superelevation is involved. The main issue I see is how `r_hat` becomes much complex and how that propagates.

I will be working based on this the following days. Any pointer, thought or suggestion is more than welcome.