RobotLocomotion/drake #7197: Organization of examples - what belongs in examples?

SeanCurtis-TRI, 2017-10-05:
Some confusion/inconsistency has crept into the `drake/examples` directory. A fuller discussion of the issue can be found [in PR 7061](https://reviewable.io/reviews/robotlocomotion/drake/7061#-). But the summary is as follows:

- Examples are largely `example/{robot|system}`
- There are *some* examples that are more `example/feature`

Left alone, a user would have to infer what the might find in `examples` and there is no clear pattern.

What needs to be done:

- We need a clear policy for examples. Do feature-based examples belong in examples? Are they merely tests?
- Resolve the code base based on the new policy (documenting and moving code as necessary).

jamiesnape, 2017-10-05:
There is also [drake-shambhala](https://github.com/RobotLocomotion/drake-shambhala) to add to the choice of locations for examples.

sherm1, 2017-10-11:
It seems natural to me that some examples would be of applications ("a wheeled robot") and others of features ("writing an event handler", "performing IK"). I think the main feature of "examples" vs. "tests" is the intended audience, with examples being designed to be useful to end users.

naveenoid, 2017-10-11:
Perhaps we can leave examples as it is and then move the robot type specific libraries etc into some other location? Such as /manipulation/kuka_iiwa_arm or /manipulation/kinova_jaco_arm. 
This way examples remains exactly what it says on the tin, a bunch of examples on how drake can be used..That way in /examples/robot_name/ I expect to see simply demo code related to how that robot can be made to do stuff in drake..

RussTedrake, 2017-10-11:
@naveenoid — in my mind, the idea behind having a manipulation directory in the first place was to capture the components that are relevant to multiple robots/examples.  Same for perception, etc.

I guess i agree with @sherm, so long as we keep the directory organized / named well.  I think this is all consistent — in my mind anything specific to eg kuka is really only an example; core drake libraries are the components that are more general tools.

SeanCurtis-TRI, 2020-05-14:
Closing this as a conversation that hasn't moved in 2.5 years.

closed 2020-05-14 by SeanCurtis-TRI