RobotLocomotion/drake #7205: Events and "relative" timed events

SeanCurtis-TRI, 2017-10-06:
There's an issue with the event system that `DrakeVisualizer` currently exposes. This is a collision between a known hack in `DrakeVisualizer` and the implicit implementation of the event system.

Description of Issue
===============

1. `DrakeVisualizer` needs to publish a "load robot" message prior to dispatching pose LCM messages.
2. There is no explicit "initialize".
3. The hack: 
    a. `DrakeVisualizer` includes a boolean value in its discrete state indicating whether the load message has been sent.
    b. In `DoCalcNextUpdateTime` it tests this discrete variable and, if still `false`, adds a `kTimed` discrete event to the event collection for some future time (currently encoded as `context.get_time() + 0.0001`).
    c. In `DoCalcDiscreteVariableUpdates` the load robot message is sent and the discrete state is set to `true`.
4. The conflict
    a. Simulator determines the duration of the next time step by polling the associated system for its next time step.
    b. The smallest time step (strictly greater than zero) is taken. And then we repeat the simulator loop.
    c. The problem is that if the discrete time step (Δt = 1e-4) is *not* the smallest, the simulator doesn't advance to a point where discrete events are handled. However, the *old* events are thrown out entirely. `DrakeVisualizer` is queried again and it submits an event always 1e-4 seconds in the future.

Possible Solutions
==============

Below, I've added a grab-bag collection of possible ways of addressing the solution. The list doesn't represent a coherent plan but merely brainstorming of related actions related to the underlying issue.

1. We are obviously hacking an alternative to having an "initialization" stage. Or having a "now" event (#5725 is specifically cited in the code.) Either of these would mitigate this particular problem.
2. We need to extend documentation prohibiting "relative" timed events as there is no guarantee they will ever be executed.  (Or we need to re-evaluate how we handle these timed events).
3. We add an "persistent" timed event that a system can dispatch once, relying that it will be evaluated without ever revisiting the logic that generated.

siyuanfeng-tri, 2017-10-06:
The current timed event logic only passes the most immediate events upstream to the simulator, and throwing away all the rest. The framework is relying on repeatedly calling CalcNextUpdateTime to get those more into the future events back. 
If we only schedule a event once, it will be lost if there are anything before it.

sherm1, 2017-10-10:
Because event handlers can make arbitrary state changes, we can't know that some event that was previously scheduled should still occur, given the new state. That's why we call CalcNextUpdateTime() before every step and never save the possibly-invalid results from a previous call. Periodic events *could* be preserved since event handlers can't change time, but they are cheap and we just recalculate them. That logic should never cause an event to be missed, however.