RobotLocomotion/drake #7215: never_destroyed constructor with zero arguments returns mysterious error message on clang

RussTedrake, 2017-10-07:
```
      static never_destroyed<std::uniform_real_distribution<double>> uniform;
```
works fine, but 
```
      static never_destroyed<std::uniform_real_distribution<double>> uniform();
```
returns 
```
error: function declared in block scope cannot have 'static' storage class
``` 
i lost some time this morning to trying to figure out why my scope was wrong.  not sure if we can do anything to help with this, but it seemed very mysterious to me.

soonho-tri, 2017-10-07:
I think the second case is interpreted as a *function declaration* (of type `void -> never_destroyed<std::uniform_real_distribution<double>>`), and C++ compiler generates an error. It is one of the places where the use of curly braces is recommended:
```
      static never_destroyed<std::uniform_real_distribution<double>> uniform{};
```

jwnimmer-tri, 2017-10-09:
Yeah, not much we can do within Drake.  In modern C++ the general rule of thumb is to use `{}` for initialization and construction, and so only use `()` for function calls.  I think that habit is the best way to mitigate this kind of problem.

closed 2017-10-09 by jwnimmer-tri