RobotLocomotion/drake #7227: LCM message generation errors during docker container build: "setup/docker/Dockerfile.ubuntu16.04.opensource"

samzapo, 2017-10-10:
Running command:
`user@hostname:/path/to/drake-distro$ docker build -t drake -f setup/docker/Dockerfile.ubuntu16.04.opensource .`

Leads to an error in the build step.

These errors are summarized by the following:
- `'//drake/*.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.`
- `ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:drake/*.py' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py'.`

```
Step 5/6 : RUN cd /drake-distro && bazel build //tools:drake_visualizer && bazel build //drake/examples/acrobot:acrobot_run_passive
 ---> Running in 5ab3d8dcffc9
Extracting Bazel installation...
...........
____Loading package: tools
____Loading package: @bazel_tools//tools/jdk
____Loading package: @local_config_xcode//
____Loading package: @local_jdk//
____Loading package: @bazel_tools//tools/cpp
____Loading package: @local_config_cc//
____Loading complete.  Analyzing...
____Loading package: drake/bindings
ERROR: /drake-distro/drake/lcmtypes/BUILD:323:1: duplicate keywords 'hdr_dest', 'guess_hdrs', 'rename' in call to install.
____Loading package: @optitrack_driver//lcmtypes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 58,969 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 164,441 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 338,521 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 607,833 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 938,585 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 1,295,961 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 1,791,577 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 2,390,617 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 3,028,569 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 3,698,265 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 4,690,521 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 5,543,513 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 6,814,297 bytes
____Downloading https://drake-packages.csail.mit.edu/director/dv-0.1.0-173-g6e49220-qt-5.5.1-vtk-8.0.1-xenial-x86_64.tar.gz: 8,067,673 bytes
____Loading package: @drake_visualizer//
____Loading package: drake/examples/zmp
____Loading package: drake/examples/kuka_iiwa_arm
____Loading package: drake/examples/pendulum
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_acrobot_u.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_acrobot_x.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_acrobot_y.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_bicycle_car_parameters_t.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_bicycle_car_state_t.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_body_acceleration.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_body_motion_data.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_body_wrench_data.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_constrained_values.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_contact_info_for_viz.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_contact_information.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_contact_results_for_viz.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_desired_body_motion.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_desired_centroidal_momentum_dot.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_desired_dof_motions.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_desired_robot_motion.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_drake_signal.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_driving_command_t.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_endless_road_car_config_t.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_endless_road_car_state_t.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_endless_road_oracle_output_t.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_euler_floating_joint_state_t.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:lcmt_foot_flag.lcm' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py_lcm_library_gen'.
[... cut ...]
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:drake/lcmt_zmp_data.py' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py'.
ERROR: /drake-distro/drake/lcmtypes/BUILD:253:1: Target '//drake/lcmtypes:drake/__init__.py' contains an error and its package is in error and referenced by '//drake/lcmtypes:lcmtypes_py'.
ERROR: /drake-distro/tools/BUILD:20:1: Target '//drake/lcmtypes:lcmtypes_py' contains an error and its package is in error and referenced by '//tools:drake_visualizer'.
ERROR: Analysis of target '//tools:drake_visualizer' failed; build aborted.
```

jwnimmer-tri, 2017-10-10:
FYI I suspect the first error is the important one:
```
ERROR: /drake-distro/drake/lcmtypes/BUILD:323:1: duplicate keywords 'hdr_dest', 'guess_hdrs', 'rename' in call to install.
```

I don't immediately know what that line is failing, though.

jwnimmer-tri, 2017-10-10:
@samzapo I'm running that command locally, and it's gotten well into the build for me, without errors.  I think the errors you posted would happen relatively immediately?

Were you building `master` at the time, and do you know which commit?

Is there any chance you were running a non-master version of the code, perhaps with a merge error within the `drake/lcmtypes/BUILD` file?

jwnimmer-tri, 2017-10-10:
My `docker build` of the most recent master 2dcfba40eebd5e5c3dcdbd081b3a379a3ff582bb just finished without any problems.  If the problem is still happening for you, I'll need more information to diagnose it.

samzapo, 2017-10-18:
The docker container builds successfully after pulling the latest master branch: https://github.com/RobotLocomotion/drake/commit/94d401975acf01443e743a3782974213c506d93a

closed 2017-10-18 by samzapo