RobotLocomotion/drake #7230: Trying to use drake-built protobuf library has some missing symbols

clalancette, 2017-10-10:
What I'm trying to accomplish is to use the version of libprotobuf (and protoc, libprotobuf_lite, etc) built and installed by drake in an outside project.  I'm using the following command to install all of drake:

```
bazel run //:install /path/to/install
```

In the external cmake project (https://bitbucket.org/ignitionrobotics/ign-msgs), I'm pointing the cmake build system at this installation target by using:

```
cmake .. -DCMAKE_INSTALL_PREFIX=/path/to/install
```

There are a couple of problems revolving around pkg-config and the name of the library that are currently causing this to fail, which I've worked around for now (I'll open PRs for those later).  However, I am running into a further issue which I'm quite not sure how to solve.  With my workarounds, I get far enough to start compiling code, but then linking fails with messages like:

```
/home/ubuntu/delphyne_ws/install/lib/libprotobuf.so: undefined reference to `google::protobuf::internal::ExtensionSet::GetUInt64(int, unsigned long) const'
/home/ubuntu/delphyne_ws/install/lib/libprotobuf.so: undefined reference to `google::protobuf::io::ArrayInputStream::ArrayInputStream(void const*, int, int)'
```

(there are a lot more of these, but I just included the first couple).  If I take a harder look at `ExtensionSet::GetUInt64`, I see something interesting.  The source file that contains the implementation (extension_set.cc) is indeed being compiled by the drake `@protobuf//:protobuf` rule, but it is never being linked into the resulting libprotobuf.so/.a .  Thus, the above link errors.  I'm not sure whether this is a problem in the protobuf rule from upstream Bazel, or in the way drake is using the rule, but I'm looking for any thoughts/advice on how to move forward on this.

clalancette, 2017-10-10:
@nuclearsandwich @stonier FYI

jwnimmer-tri, 2017-10-10:
That symbol should be coming from `protobuf_lite`; the cps file doesn't declare that `protobuf` depends on `protobuf_lite`, but my quick skim of protobuf's upstream BUILD files suggests to me that it should depend.  Maybe some fiddling around there would fix it up?

Bigger picture, perhaps we should perhaps compare Drake's install artifacts related to protobuf, versus what upstream's build system on its own would install (via autotools, I think?).  To the extent those results differ from Drake's, we open ourselves to problems like this.

clalancette, 2017-10-16:
Given the offline discussions we had, I think we concluded that we are going to try to shoot for using the system protobuf.  Thus, I am going to close this out as it isn't really that important if we are using the system version.  Feel free to re-open if you think otherwise.

closed 2017-10-16 by clalancette