RobotLocomotion/drake #7240: Obtain SNOPT source archive from other than git

jwnimmer-tri, 2017-10-11:
I have never been a fan of Drake obtaining snopt via a git clone -- its slow, brittle, uses out-of-band authentication, etc. If anyone has suggestions for an alternative mechanism they would prefer, I'm open to suggestions.

In http://drake.mit.edu/bazel.html#proprietary-solvers for example we have GUROBI_PATH environment variable that the developer (or eventually CMake) sets to where they have manually downloaded Gurobi. You imagine doing a similar thing with Snopt -- that Drake could be configured to compile against user-supplied Snopt source archive.

Possibly we could even offer the environment variable (or similar), but default to trying git as a fallback.

RussTedrake, 2017-10-12:
a few relevant thoughts/concerns:  
- we currently administer the snopt site license via adding people carefully to the github team "HAS_SNOPT_LICENSE".  MIT CSAIL and TRI both have site licenses.  We would need an alternative mechanism.
- a pain point with the above (which it would be great to resolve) is that external users who DO have their own snopt license are unable to use that mechanism, so are unable to build with snopt.  
- there is a limited free student license that would also be good to support if students have access.  

I think the SNOPT_PATH option might satisfy all of the above.  
One other pain point is that I believe we have not yet updated our API to the bleeding edge snopt distribution (which is the default when users download snopt from the website).  So version checking, and a helpful error message, will be required.

jwnimmer-tri, 2017-10-14:
FYI We should weight #7259 into this decision as well -- how any solution will chain into downstream, Bazel-using projects.

jwnimmer-tri, 2018-01-19:
My inclination would be to remove our `BUILD` files from the RL/snopt repo, and instead place them inside Drake like we do with most other externals.  (I already had to write those out-of-tree BUILD files, for use within Driving.)  Then as a starting point, we offer a way to use the upstream snopt source archive through some configuration mechanism (maybe an env var; I am going to consider whether a bazelrc would be better for these things too).   I also suspect this would let us support different snopt versions more gracefully.

jwnimmer-tri, 2018-01-31:
FYI Probably we should also account for #245 when designing the new mechanism.

jwnimmer-tri, 2018-03-06:
So I think #8031 was the only Drake code required to solve this issue; the remaining step is to add CI coverage, which I think will be only `drake-ci` changes.

We'll need CI to do something equivalent to `export SNOPT_PATH=/home/username/Downloads/snopt7.6.tar.gz` prior to running the Drake build.

@jamiesnape Can I kick this issue over to you to finish up?  Please feel free to ping me if something is wrong with the code that I can fix along the way.

jamiesnape, 2018-03-06:
The link Russ was sent only had a ZIP file. I have repacked it for now, but you may want to support ZIP natively.

jamiesnape, 2018-03-06:
Also the docs refer to 7.2 in some places and 7.5/7.6 elsewhere.

jwnimmer-tri, 2018-03-06:
My intent is that #8263 will fix all snopt version-skew typos in the docs; if you notice any it missed, let me know.

jwnimmer-tri, 2018-03-06:
> The link Russ was sent only had a ZIP file. I have repacked it for now, but you may want to support ZIP natively.

Yeah, there's a TODO in the code to support zip.  I can add it, but I'd like to have CI for all code in repository.bzl.  Do you think I should drop tar.gz when I add zip, or is it plausible to have both in CI (as well as git)?

jamiesnape, 2018-03-06:
Plausible, but we may have to have separate jobs and not sure we want to go down that route just to exercise two lines of code. If you can write unit tests that are maybe opt in for CI that exercise the other download methods, then we could do that, otherwise I would rather we just support one method in CI.

jwnimmer-tri, 2018-03-06:
Makes sense.  I will take this back while I have a think and check on `*.zip` support.

jwnimmer-tri, 2018-05-11:
@jamiesnape I am fine to replace targz support with zip support instead.  I don't need either one in-house, as it turns out.  Can I kick it over to you to work on that replacement, and then adding zip testing to CI (while also keeping git testing intact)?

Note #8801 where I also plan to edit the `repository.bzl`.

EricCousineau-TRI, 2018-12-03:
Closing, as this seems like it's been resolved.

closed 2018-12-03 by EricCousineau-TRI

jwnimmer-tri, 2018-12-03:
To my knowledge, we don't have any CI builds that set SNOPT_PATH=/path/to/archive.tgz (or zip).  We should leave some issue open (or file a new issue) until we have Jenkins coverage for that.  Probably #8869 can be that issue, so we can leave this one closed.