RobotLocomotion/drake #7251: Add to Python RigidBodyTree API to enable collision-free path planning in dynamically changing scenes

EricCousineau-TRI, 2017-10-12:
Per [this request](https://github.com/RobotLocomotion/drake/issues/7185#issuecomment-336192337) in #7185:
Before the features in #7025 / #5013 are completed for `MultibodyTree`, this could be achieved by updating the existing Python bindings (adding `addCollisionElement`, `removeCollisionElement`, `removeCollisionGroupsIf`, `updateCollisionElements`, and the upstream classes for that API).

Per f2f with @avalenzu, this could be achieved by adding the "scene" collision elements to the world body with a unique group name, and when the scene is updated (either with new objects, or updating the poses of the existing elements), either update the element position or remove the existing group and the updated elements anew.

\cc @dmsj

jwnimmer-tri, 2018-12-21:
@EricCousineau-TRI I'm assuming that we don't care about new RBT features anymore?  Can this be closed?

EricCousineau-TRI, 2018-12-21:
Yup. We have other issues tracking this equivalent for MBP.

closed 2018-12-21 by EricCousineau-TRI