RobotLocomotion/drake #7252: Cleanup using directives from `free_body.h`

amcastro-tri, 2017-10-12:
A number of `using` directives in header for benchmark `free_body` should be removed.
See [comment here](https://reviewable.io/reviews/robotlocomotion/drake/7218#-KwGLEph4LDAXEu3oLab).

amcastro-tri, 2017-10-12:
I'll transcribe @soonho-tri's question from [comment here](https://reviewable.io/reviews/robotlocomotion/drake/7218#-Kw0c1xKWbFGSkZPCUBV) so that it doesn't get lost:
> Can you provide a comment why we have this particular constant here? (and also for get_J below) For example, are they from the book cited above?

jwnimmer-tri, 2019-06-20:
> A number of using directives in header for benchmark free_body should be removed.

This is already done.  I've re-titled this issue to focus on the second post, instead.

mitiguy, 2019-07-03:
@jwnimmer-tri   
Jeremy - can we close issue #7252 
(Justify the get_i and get_j constant values in free_body_benchmark) ?

jwnimmer-tri, 2019-07-03:
The fix I made was only to repair the using statements.
To my knowledge, no change has been made to fix the documentation of the constants.

jwnimmer-tri, 2024-03-16:
Well, the free_body docs still don't explain where `I` and `J` came from.  The next time someone decides they need to know about it, they can work to dig up the answer and add it to the docs.

closed 2024-03-16 by jwnimmer-tri