RobotLocomotion/drake #7253: UBSan failure in lcm:drake_lcm_log_test

m-chaturvedi, 2017-10-12:
To reproduce:
`bazel test --config=ubsan //drake/lcm:drake_lcm_log_test //drake/systems/lcm:lcm_log_playback_test`

This was exposed in LCM after #7178 . [LCM Issue](https://github.com/lcm-proj/lcm/issues/194).  [Proposed PR in LCM](https://github.com/lcm-proj/lcm/pull/195).

jwnimmer-tri, 2017-10-12:
I am skeptical of the PR proposed in the link.  Isn't the correct fix to use `uint32_t` bit-mixing magic instead, not disable the warning?  The warning seems like a true positive to my first reading.

m-chaturvedi, 2017-10-12:
You're right.  From their docs:
>LCM Sync Word is an unsigned 32-bit integer with value 0xEDA1DA01

[And they have this](https://github.com/lcm-proj/lcm/blob/master/lcm/eventlog.c#L18).

m-chaturvedi, 2017-10-16:
Closed via https://github.com/RobotLocomotion/drake/pull/7266

closed 2017-10-16 by m-chaturvedi