RobotLocomotion/drake #7260: Need CI coverage of `bazel run :install` correctness

jwnimmer-tri, 2017-10-14:
Drake's CI should have coverage that `bazel run :install` behaves correctly, in terms of a CMake project being able to consume the results.  (I'm leaning toward pre-merge checks, though possibly `Continuous Production` that we could opt-in to for `Experimental` would be good enough.)

The fix in #7257 for a bug that I introduced in #7241 is a good example of what we would like to catch, since it broke ~everything.  Other more subtle or rare failure modes perhaps don't need to be covered, at least in the first iteration.

Possibly #7198 will get us most of what we want.

EricCousineau-TRI, 2018-05-03:
@jamiesnape Would you mind if I assigned this to you?

For CMake project consumption within Drake's CI, should we make a simple non-cached test which is simply a *very* small subset of `drake-shambhala/drake_cmake_installed`, consuming from `bazel run //:install`, setting the appropriate compilers per Bazel's setup?

Or do we want want this to run entirely through the CMake install process? (in which case, would we just want to defer this entirely to `drake-shambhala`?)

jamiesnape, 2018-05-03:
> @jamiesnape Would you mind if I assigned this to you?

Sure. Not decided on how we do it, but I don't think it will be deferring to shambhala.

jwnimmer-tri, 2018-11-14:
The drake-shambala CI is good enough as a first pass.  If we discover that it's not catching certain kinds of errors, we can open separate issues for those problems.

closed 2018-11-14 by jwnimmer-tri