RobotLocomotion/drake #7305: Extend maliput API with `RoadRuleset` to capture rules of the road

maddog-tri, 2017-10-19:
Here is a first pass at a design for `RoadRuleset`, a (long-awaited) addition to the `maliput` API for obtaining information about "rules of the road" features on a road network.  The immediate goal of this issue is to capture a transcription of notes from my notebook and to make them available for comment from a wider audience.

The features/rules envisioned at the moment (roughly in priority order) are:
  * direction/usage
  * stop lines
  * speed limits
  * passing restrictions
  * ongoing route ("turning") restrictions
  * vehicle restrictions

`RoadRuleset` will be loosely coupled to `RoadGeometry` via entity ID's (`RoadGeometryId`, `LaneId`, etc).  There will be no direct memory/pointer references between the two.

The basic query will have the form:
> Return all rule elements affecting a specified `LaneId` which are effective during the time interval `[t0, t1)` and within the longitudinal range `[s0, s1]`, potentially limited to specified rule types.

Filtering by `SegmentId` or `JunctionId` would be useful as well (e.g., "Tell me speed limits of all adjacent lanes in my current segment.").

The return value is a list of rule elements.  Every element has the common fields:
  * `LaneId`
  * time interval `[t0, t1)`

### Element subtypes

#### Direction/Usage
  * longitudinal extent `[s0, s1]`
  * rule/designation --- 1 of:
    * unspecified/bidirectional
    * +s only
    * -s only
    * bidirectional turning lane
    * restricted/no-traffic
  * _Should "no applicable record found" equate to "unspecified"?_
  * _More possibilities:  parking, pedestrian, bike-lane,..._

#### Stop Line
  * longitudinal location `s`
    * _(Maybe should be an extent `[s0, s1]`.  E.g., consider a crosswalk as
      one stop-line or two?)_
  * signal type --- 1 of:
    * yield sign
    * stop sign
      * 1-way
      * N-way (with reference to adjacent signs?)
      * _(Or, is N-way to be determined by examining junction geometry?)_
    * stop-light
      * unique-id
      * _Should static rules be implemented via `[t0, t1)` intervals?_
        _Or should all stop-lights' state be via Id reference to a
        controller object living somewhere else?_
    * crosswalk?
  * applicable direction
    * "facing -s"
    * "facing +s"
    * "both directions"

#### Speed Limits
   * longitudinal extent `[s0, s1]`
   * limit value (speed)
   * limit type:
     * strict
     * advisory
   * applicable vehicle:
     * any
     * trucks
     * ?

#### Passing Restrictions
  * longitudinal extent `[s0, s1]`
  * direction
    * to-left
    * to-right
  * rule:
    * lane change allowed ("white dash")
    * avoid lane change ("white solid")
    * pass with care ("yellow dash")
    * no passing ("yellow solid")
    * exit ("short white dash")

#### Ongoing Route ("Turning") Restrictions
  * which end (kStart, kFinish)
  * ongoing LaneEnd (LaneId, WhichEnd)
  * restricted vehicle type
    * (not) any
    * (not) bus
    * (not) truck
    * ...
  * _(Or, maybe this concept is better represented by vehicle restrictions
    on the ongoing lane instead.)_

#### Vehicle Restrictions
  * longitudinal extent `[s0, s1]`
  * vehicle type
    * HOV only
    * no trucks
    * bus only
    * etc

maddog-tri, 2017-10-19:
@jwnimmer-tri @stonier

jwnimmer-tri, 2017-10-26:
One suggestion: I would hesitate to move too far along with the above ideas, unless we concurrently design for how the visual / physical representation of them is denoted.  In particular, especially the paint road surfaces (lines, arrows, etc.) and the traffic control elements (lights, signs).  I suspect we will want correlation between their semantics and their visual / physical representations, and so I'd like to see at least a first pass sketch of how the above data set relates to the visual / physical dataset alongside it.

liangfok, 2018-01-04:
Regarding:

> Should static rules be implemented via [t0, t1) intervals? Or should all stop-lights' state be via Id reference to a controller object living somewhere else?

I think it should be the latter. In other words, Maliput itself should not be a simulator.

agalbachicar, 2018-01-08:
A few comments / thoughts:

--------
> Stop Line
>  - longitudinal location `s`
>    - (Maybe should be an extent `[s0, s1]`. E.g., consider a crosswalk as one stop-line or two?)

An alternative could be `[s, s_min, s_max]` where the last two can be optional.

--------
> Stop Line
>  ...
>   - signal type --- 1 of:
>    ...
>    - stop sign
>      - 1-way
>      - N-way (with reference to adjacent signs?)
>      - (Or, is N-way to be determined by examining junction geometry?)

I guess `N-way` should be enough. I think both `N-way to be determined by examining junction geometry?` and `(with reference to adjacent signs?)` are not aligned with the type of queries shown below (properties of a lane or a lane-collection vs. lane X against lane Y).

--------
> Stop Line
>  ...
>  - direction usage

Is it necessary? Isn't it covered by `Direction/usage` property?

liangfok, 2018-01-11:
Here are my detailed comments:

# Comment 1: Concern About Inclusion of Time Interval in Rule Element

I'm concerned about including time interval `[t0, t1)` in a rule element because that forces Maliput to predict what's going to happen in the future and thus become a simulator. Since the time interval is continuous, Maliput must be able to step time forward by arbitrarily small increments. Specifying the rules of the road in a sufficiently fine granularity or in a continuous manner seems non-trivial, and may result in an explosion in the number of rule elements returned by a query if the time interval of each rule element is very small. Consider keeping Maliput a data structure and repository that can be affected by external scenario-specific agents, and only returning the *current* rule set.

Maybe the query can be the current geo position and a lookahead distance. This would allow rule elements for more than one LaneId to be returned.

# Comment 2: Concern About How to Express Rules that Span a Sequence of Lanes

For example, how do we express that a sequence of lanes is part of a toll road? Maybe a rule element should contain a set of LaneIds?

# Comment 3: Consider Separating Lane Direction From Usage

Combining lane direction and usage into a single rule element is problematic because direction and usage are not tightly coupled (e.g., a single lane direction may be used for many purposes). Figuring out all combinations of them may be confusing. I think the following directions should be initially provided:

1. unspecified (default)
2. bidirectional
3. +s
4. -s
5. restricted / no traffic

Usage details like whether it's "bidirectional" vs. "bidirectional turning" can be omitted and obtained through other means like looking at the road geometry. Usage details like whether the lane is for parking, pedestrians, bicycles, can be obtained via vehicle restriction rule elements as additional vehicle types.

# Comment 4: Stop Line: No Need for N-Way Stop Sign Rule Element

I vote for not providing N-way stop signs as a rule element since agents can determine that based on the  junction that should hold all of the stop signs.

# Comment 5: Stop Line: No Need for Crosswalk Rule Element

A crosswalk rule element is the same as a yield rule element.

# Comment 6: Stop Line: Should Include Width

I think stop line should be specified by an `[s_min, s_max]` extent. Can we assume that no stop lines cross lane boundaries?

# Comment 7: Speed Limits Rule Elements Should Include Additional Details

Additional details like speed limits that only apply to trucks above a certain weight, or during daylight vs at night.

# Comment 8: Remove Ongoing Route Restrictions

Let's put this info in vehicle restrictions on ongoing lane and have the user query for it.

# Comment 9: Vehicle Restrictions Rule Element Should Contain Vehicle Properties

This in in addition to vehicle type. For example, vehicle height, weight, and whether it's holding hazardous waste (and what type).

# Comment 10: Should a Rule Element Contain General Instructions or Warnings?

For example, "watch for children playing", "frequent dear crossing", "low salt area" / "no spray zone", "thickly settled" (something I first encountered in New England), "bridge may ice over before road", "blind driveway", "steep grade", "no engine braking", "falling rocks", "Amish buggy ahead", "metal plates on ground", "uneven road surface", "freeway crossing", "open range", "detour", "exit", etc.

# Comment 11: Consider renaming "Passing Restrictions" to "lane change restrictions"

This is to cover scenarios where we want to pull into a parallel parking spot that's next to the current lane.

liangfok, 2018-01-19:
Just to clarify, the proposal above is for the **runtime** API only. It's a description of what agents can query.

liangfok, 2019-06-08:
Closing since we've made significant progress in this area and future progress can be tracked by more narrowly focused tickets in Maliput's dedicated downstream repo.

closed 2019-06-08 by liangfok