RobotLocomotion/drake #7306: Update CI images for new protobuf2 dependencies

jwnimmer-tri, 2017-10-20:
PR #7303 added some new OS dependencies.  Please update CI's default Xenial and Apple images to provide the new packages by default.  Thanks!

jamiesnape, 2017-10-20:
Do we have any other system dependency updates in the pipeline?

jwnimmer-tri, 2017-10-20:
I was wondering the same thing.  I didn't immediately think of any, but I'll do another survey to be sure.

jwnimmer-tri, 2017-10-20:
I guess #7293 has some, though perhaps they are all already present in CI.

jamiesnape, 2017-10-20:
Yes, they are. They are really refactoring from the manual instructions to the script.

jwnimmer-tri, 2017-10-20:
On Xenial, it looks like `libyaml-cpp-dev` will be coming down the pipeline eventually too.  I suspect that `brew install yaml-cpp` might also apply in that case, though I'm less sure there.

jamiesnape, 2017-10-20:
Version numbers of the Homebrew and Xenial packages are pretty similar, so I imagine so.

jamiesnape, 2017-10-20:
Mind if we PR the yaml-cpp updates to `install_prereqs.sh` now? Keeps (practically) everything that is installed on CI transparent.

jwnimmer-tri, 2017-10-20:
SGTM.

BetsyMcPhail, 2017-10-20:
CI images have been updated to match latest install_prereqs scripts.

jwnimmer-tri, 2017-10-23:
@clalancette Can you verify and then we can close this issue?  Maybe rebase and push #7278?

clalancette, 2017-10-23:
Will do.

clalancette, 2017-10-23:
I'll note, though, that I'm sure MacOS is going to fail with that branch.  I still haven't figured out how to make it work properly.  That being said, it will still be a good test for Linux.

clalancette, 2017-10-24:
So, despite the fact that one of the builds in #7278 failed, I think we can call this done.  That failure was a segmentation fault in two of the tests.  While that is obviously a problem for that PR, it doesn't seem to be a problem with the setup/build portion of it.

jwnimmer-tri, 2017-10-24:
Great.  Thanks @BetsyMcPhail!

closed 2017-10-24 by jwnimmer-tri