RobotLocomotion/drake #7314: kuka examples are broken

RussTedrake, 2017-10-23:
After partial repairs in #7313, I still see the follow set of affairs in my kuka examples directory:
- state_lcm_translator is broken; director is failing to load the IIWA_STATUS.  possibly due to #7166?
- kuka_sim_dual.pmd can't even run the simulation.  this example should be fixed or dropped.  if fixed, please add to the README.md started in #7313.

How do we put better test coverage on these?

sammy-tri, 2017-10-23:
> state_lcm_translator is broken; director is failing to load the IIWA_STATUS. possibly due to #7166?

I suspect it is due to #7166 and the fact that any directorPython you managed to load from `drake-distro/build` would be from back in the CMake days and wouldn't be getting lcmtypes updates.

> kuka_sim_dual.pmd can't even run the simulation

Will fix and add a test.

sammy-tri, 2017-10-23:
Opened #7317 to fix the dual arm simulation.

sammy-tri, 2017-10-25:
In light of the issue I just noted in #7204 (our prebuilt drake-visualizer can't load robot models), reviving director_ik_app.py is effectively impossible (which, uh, reduces the importance of the state translator).

Given this state of affairs, I'm not totally sure what the best thing to do is.  One idea would be to create a couple of canned demos which send a canned plan to a simulation (probably single arm and dual arm versions) either by doing some IK or just recording the series of joint states and sending a trajectory.  Does that sound worthwhile?  Other ideas?

RussTedrake, 2017-10-26:
if the pick and place demos are coming soon from @avalenzu , then perhaps they can play that role?
i do think having a graphical interface to IK is highly desirable.

N.B.  - @sammy-tri  i think you might have linked to the wrong issue number above?

sammy-tri, 2017-12-14:
This came up in https://github.com/RobotLocomotion/drake-iiwa-driver/issues/14#issuecomment-351828641 -- without a working directorPython built against drake, providing a graphical IK example is going to be tough.  This may call for increasing the priority of #7024 (built against a fixed drake target) or finding a way to package directorPython along with the files currently in `examples/kuka_iiwa_arm` to deliver an example which can run alongside a binary install of drake.

EricCousineau-TRI, 2018-12-10:
Just to check, are these issues still relevant? This is currently marked as high-priority, would be nice to either lower the priority or close the issue if possible.

sammy-tri, 2018-12-10:
I think it's up to Russ to determine if the manipulation station examples are sufficient to close this ticket (I'm leaning toward "yes" on that question).  I'll lower the priority at any rate.

sammy-tri, 2018-12-10:
I guess I should elaborate on my proposal for closing, since I don't think I mean just pressing the "close" button:

Resolve this and #7321 by removing all director-based examples and any other broken stuff.

PR #10655 (2019-02-12): Remove the director-based iiwa examples.
These haven't worked in a long time since we haven't had a version of
director with the needed functionality available.

Fixes #7314

Fixes #7321

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10655)
commit be778520 (2019-02-12): Remove the director-based iiwa examples.

These haven't worked in a long time since we haven't had a version of
director with the needed functionality available.

Fixes #7314

Fixes #7321
closed 2019-02-12 by sammy-tri