RobotLocomotion/drake #7321: remove directorPython usage from iiwa arm example

RussTedrake, 2017-10-24:
As discussed in https://reviewable.io/reviews/RobotLocomotion/drake/7313#-Kx99MsvVKtz09Qu5tIr

Nominal plan: convert the state translator to C++ (and cover it with unit tests), and to convert the director ik interface to a drake-visualizer app if necessary.

Related: #7024

sammy-tri, 2017-10-24:
Also the kuka example's README.md should be updated to include the steps which it's currently missing which depend on this issue (making sure to also add the dual-arm examples).

sammy-tri, 2017-12-11:
I've got a C++ version of the state translator basically ready (I think I might wait on #7439 before opening a PR).  As we've discussed in other places (#7314 and #7024) converting the director IK interface to a drake-visualizer app requires a more complex resolution.  Do we want to leave this issue open pending that change?  Or close this and just track the lack of graphical IK in #7314?

PR #10655 (2019-02-12): Remove the director-based iiwa examples.
These haven't worked in a long time since we haven't had a version of
director with the needed functionality available.

Fixes #7314

Fixes #7321

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10655)
commit be778520 (2019-02-12): Remove the director-based iiwa examples.

These haven't worked in a long time since we haven't had a version of
director with the needed functionality available.

Fixes #7314

Fixes #7321
closed 2019-02-12 by sammy-tri