RobotLocomotion/drake #7331: Refactor install.py.in

fbudin69500, 2017-10-25:
`install.py.in` is not tested. It is getting complex and it is complicated for reviewers to verify that modifications do not break anything.

Fixup methods should be refactored and have unit tests as well as pycodestyle checking. Those function need to turn into *.py code that is imported by the generated :install program(s).

See comments on PR #7301

jamiesnape, 2017-11-15:
Relates #6117.

jwnimmer-tri, 2018-11-14:
(Yes; see 6117 for a possible nit to look into as part of this.)

EricCousineau-TRI, 2018-12-08:
#10103 might be able to address simplified testing.

rpoyner-tri, 2021-12-13:
Unless someone else has work-in-progress on this, I will take a look.

jwnimmer-tri, 2021-12-20:
In #16252, I moved the `install.py.in` code into an `installer.py` binary, which automatically enabled our default linting checks for it.

What remains here is to refactor `installer.py` for maintainability and test coverage.

jwnimmer-tri, 2022-05-06:
As much as I still don't enjoy the status quo of `installer.py` code, I think refactoring for testability and maintainability is best done _in situ_ as we fix bugs our add new features, not as a standalone task.

closed 2022-05-06 by jwnimmer-tri