RobotLocomotion/drake #7335: Use system nlopt

soonho-tri, 2017-10-25:
For now, we're building [nlopt](https://nlopt.readthedocs.io/en/latest) in bazel build. I'm wondering if we can switch to use the one in a system. FYI, both of macOS (via homebrew) and Xenial provides the latest version of nlopt, `2.4.2`.

 - Homebrew: https://github.com/Homebrew/homebrew-core/commits/master/Formula/nlopt.rb
 - Xenial https://packages.ubuntu.com/source/xenial/nlopt

/cc @jwnimmer-tri , @stonier.

soonho-tri, 2017-10-25:
My motivation is that dReal has a plan to use (system's) `nlopt` and I'd like to avoid any possible conflicts in Drake (between the system nlopt and locally-built nlopt).

Also `nlopt` is under LGPL which requires dynamic linking. I think relying on the system one can prevent possible static linking (or not?)

soonho-tri, 2017-10-26:
@clalancette , could you check if this change can cause a problem (vs. ROS)? @jwnimmer-tri shared  a document, `Dependencies Drake vs ROS (Xenial)` with me, and it seems that we won't have a problem.

clalancette, 2017-10-26:
@soonho-tri I did a bit of investigating here.  You are absolutely correct; Xenial does have nlopt 2.4.2, so that will be compatible with what is in Homebrew (it also happens to be the latest official release from nlopt).  2.4.2 is older than what drake is currently using, but from a brief look of the changes since 2.4.2, none of them seem to be API-breaking, so that will be OK as well.  As far as linking with ROS, I don't forsee any trouble; a bunch of other ROS packages depend on the system version of nlopt.  So it seems like it would be totally fine to change to a system version of nlopt.  I hope that is helpful.

soonho-tri, 2017-10-26:
@clalancette , thanks a lot for the detailed analysis. I'll open a PR making the transition.

PR #7386 (2017-11-02): Use system nlopt
Close #7335

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7386)
closed 2017-11-02 by soonho-tri