RobotLocomotion/drake #7351: Building without Matlab

joaospinto, 2017-10-27:
After cloning and installing the pre-requisites, I create a `build folder` inside `~/github.com/RobotLocomotion/drake`, and run `cmake .. && make -j$(nproc)` from within that folder. I also tried to run `cmake -DDISABLE_MATLAB=ON ..` instead of `cmake ..`. My build always fails with the following error:
`CMake Error at /home/joao/github/RobotLocomotion/drake/cmake/matlab.cmake:66 (matlab_add_unit_test):
  Unknown CMake command "matlab_add_unit_test".`
What's the problem here? Is it possible to build your project without having Matlab installed? Thanks.

Capri2014, 2017-10-27:
@joaospinto  what system are you using ?

Capri2014, 2017-10-27:
@joaospinto Did you try build with bazel ? http://drake.mit.edu/bazel.html?highlight=bazel%20build

joaospinto, 2017-10-27:
@Capri2014 The fact that bazel does not work well inside docker makes it slightly inconvenient to me. There seem to be ways available workarounds for that, so I will look into those. Did you have the same problem when building with CMake? Or you just uesd Bazel by default?

PR #7293 (2017-11-02): Use Drake built with Bazel for MATLAB bindings
Closes #6728, closes #7328, closes #7341, closes #7351, and closes #7375 .

This will need some pretty major CI changes before merging. There is a lot of documentation that still needs updating and this is by no means the final version of MATLAB support in Drake. It defines several CMake options: `WITH_GUROBI`, `WITH_MATLAB`, `WITH_MOSEK`, `WITH_SNOPT`. Consider  `WITH_MATLAB` and `WITH_SNOPT` fully functional. The others may need some finesse to get working. This is only tested with the `Unix Makefiles` generator so far. There is probably little  [... cut]
closed 2017-11-02 by jamiesnape