RobotLocomotion/drake #7355: Consider providing `pkg-config` files (especially for `eigen3`)

EricCousineau-TRI, 2017-10-27:
Since some incantations of `FindEigen3.cmake` use `pkg-config`, we may want to consider packaging `pkg-config` files using Matt's CPS setup.

jamiesnape, 2017-10-30:
I imagine we will be switching to the system version of Eigen3 at some point. For the rest of Drake, we made a conscience decision not to support `pkg-config`. Obviously that could be revisited, but supporting both Bazel and CMake is already a lot of work.

jwnimmer-tri, 2017-10-30:
> I imagine we will be switching to the system version of Eigen3 at some point.

Uncertain, but my guess would be not until Ubuntu 18.04 Bionic.

jwnimmer-tri, 2018-12-21:
FYI We will _not_ use Eigen3 from Bionic; it's not `>= 3.3.3`, which is our baseline.

jwnimmer-tri, 2019-10-30:
Actually, Bionic has `3.3.4` now.

jwnimmer-tri, 2019-10-31:
Oops, I somehow fat-fingered the assignee widget yesterday.

jwnimmer-tri, 2020-04-02:
@EricCousineau-TRI now that we use the pre-existing Eigen (not installing our own), I think this issue can be closed?

EricCousineau-TRI, 2020-04-02:
Yup!

(Also, this was written for `catkin` / ROS1-based workflows; given that ROS2 / `ament_cmake` no longer seem to generate / interface with `pkg-config`, yeah, it's way less important.)

closed 2020-04-02 by EricCousineau-TRI