RobotLocomotion/drake #7377: collision geometry for <link name="world"> not included in RigidBody

dmsj, 2017-10-31:
@RussTedrake per our conversation:

When creating a new `RigidBodyTree `for collision free IK planning, e.g. `self.robot = pydrake.rbtree.RigidBodyTree(self.config["ik_urdf"])`, it appears that `<collision>` geometry in the `<link name="world">` is being incorrectly filtered out. 

For example, with the following URDF world definition:
```xml
  <link name="world">
    <visual>
      <origin xyz="0 0 -0.005"/>
      <geometry>
        <box size="3.0 3.0 0.01"/>
      </geometry>
    </visual>
    <collision>
      <origin xyz="0 0 -0.005"/>
      <geometry>
        <box size="3.0 3.0 0.01"/>
      </geometry>
    </collision>
  </link>
```
The output of 
```c++
for (auto body_iter = bodies.begin(); body_iter != bodies.end(); ++body_iter) 
    {
        std::cerr << "body name: " << *(*body_iter) << '\n';
    }
```
correctly outputs collision geometry for other links (e.g. the iiwa) but doesn't find the collision for world:
```text
body name: RigidBody
  - link name: world
  - parent joint: no parent joint
  - Collision elements IDs: []
body name: RigidBody
  - link name: iiwa_link_0
  - parent joint: torso_to_iiwa_left
  - Collision elements IDs: [140194888682912]
body name: RigidBody
  - link name: iiwa_link_1
  - parent joint: iiwa_joint_1
  - Collision elements IDs: [140194888684416]
etc...
```
cc @mrhebert @jaymwong

SeanCurtis-TRI, 2017-10-31:
This is a known "feature".

https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/parsers/urdf_parser.cc#L500

The urdf parser explicitly skips any body named "world". I don't know the historical reason for that, but it's been that way for a long while.

liangfok, 2017-10-31:
I believe I added the special-case handling of the "world" body here:

https://github.com/RobotLocomotion/drake/pull/2171/files#diff-414acd4947c600b5bafcc5ef751dfeadR318

The idea was to use a link called "world" to enable users to specify how the rest of the robot connects to the world, i.e., the "floating base type".

dmsj, 2017-10-31:
Ha! I love "features." Thanks! Maybe a comment in the docs to help users down-the-line ;-)

liangfok, 2017-10-31:
Probably the closest doc we have now is in a .cc file:

https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/parsers/urdf_parser.cc#L946-L948

Though there is a need to remove the feature: #3081.

jwnimmer-tri, 2019-11-01:
@SeanCurtis-TRI This appears to be a request specifically related to RigidBodyTree.  Can we close it as "WONTFIX" in that case?

SeanCurtis-TRI, 2019-11-01:
We can close it.

However, my recollection is that the RBT URDF parser was ported pretty directly over for MBP. As such, I believe we have the same artifact. See [`detail_urdef_parser.cc`](https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/detail_urdf_parser.cc#L102-L104). If the body's name is "world", we skip it.

So, the issue has propagated to MBP.

jwnimmer-tri, 2019-11-01:
Maybe update the title / overview, in that case?

SeanCurtis-TRI, 2019-11-01:
Done

SeanCurtis-TRI, 2021-12-04:
I believe this was inadvertantly duplicated by #14727  and resolved by #15763.

closed 2021-12-04 by SeanCurtis-TRI