RobotLocomotion/drake #7382: Fail hard if the wrong Eigen is found

stonier, 2017-10-31:
That is, when doing find_package(drake) on an installed version of Drake.

jamiesnape, 2017-11-02:
CMake will already fail hard if it finds an incompatible version. Where are you seeing problems?

jwnimmer-tri, 2017-11-02:
From timing, I suspect it was related to RobotLocomotion/drake#7378.  (I don't have any more details, though.)

jamiesnape, 2017-11-02:
So do we 3.3.3 or 3.3.4? If we need 3.3.4, then we need to tell CMake. Otherwise #7378 should be entirely redundant unless you are doing something that you shouldn't be with CMake.

jamiesnape, 2017-11-02:
@EricCousineau-TRI do you have any background since you opened that PR?

jamiesnape, 2017-11-02:
(And if you happen to be calling `find_package(Eigen CONFIG REQUIRED)` independently and not specifying a version, then clearly you are not going to get a hard fail. The solution there is to correctly specify your requirements.)

jwnimmer-tri, 2017-11-02:
To my knowledge, Drake needs only 3.3.3.

I was happy to accept the "If we have 3.3.4, then disable the traits thing that is now provided upstream" portion (`common/eigen_autodiff_limits.h`), because it enables off-label use of Drake with newer Eigen, and would be correct when we add support for Ubuntu 18.04 in the future, for example.

As to why we are improving the diagnostic with Eigen < 3.3.3 (`common/autodiff.h`), I don't know.  I agree it seems odd given that CMake should already have complained by then, but seemed like a reasonable second-line fail-fast.  (In retrospect, we could probably improve the `static_assert`'s comment though, hinting to users that their CMakeLists.txt files are broken somehow.)

jamiesnape, 2017-11-02:
This looks suspicious, though: https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/eigen/eigen-create-cps.py#L14

jamiesnape, 2017-11-02:
I think maybe it is the definition of "incompatible version" then. The changes in #7396 fix that. As much as people thing CMake looks around their whole system for literally any version of a package, that is not how it works, so issues like #7378 should probably be brought to our attention to fix.

jwnimmer-tri, 2017-11-02:
Agreed.

jwnimmer-tri, 2018-11-09:
It seems to me like nothing is left to do here.

closed 2018-11-09 by jwnimmer-tri