RobotLocomotion/drake #7387: Using a .deb (Ubuntu) and a homebrew-bottle (macOS) locally in bazel

soonho-tri, 2017-11-01:
@jwnimmer-tri and I are working on this direction. We think that this is suitable for the dependencies which we expect frequent updates (i.e. dReal). Pros of this approach are:

 - No need to update CI images when an upstream is updated.
 - Fast build-time (compared to compile source) since it's relying on precompiled packages.

@jwnimmer-tri already wrote `new_deb_archive` at https://github.com/jwnimmer-tri/drake/blob/bazel-debs/tools/workspace/deb.bzl. I'm working on a macOS/homebrew counterpart.

/cc @jamiesnape , there is a small issue. Since I don't have a macOS 10.11 (El Capitan) machine at hand, I can't create bottles for El Capitan. For now, our plan is to disable dReal supports for El Capitan.

jamiesnape, 2017-11-01:
This would be installing into a location in the sandbox?

soonho-tri, 2017-11-01:
@jamiesnape , correct.

jwnimmer-tri, 2017-11-01:
The use case story is for dependencies that are not available via the OS, but that we don't wish to build from source all the time (e.g., because they are huge, or difficult to Bazel'ify).  The workspace would download a pre-compiled image, make it available via appropriate BUILD labels, and (if needed at runtime) install any relevant shared libraries, data, cmake metadata, etc. as part of Drake's install rules.

Challenges include making sure the build process is repeatable and available to enough Drake developers so that we can maintain it as a community, and making sure the binary artifacts are hosted reliably.

I'm relatively certain that in some cases, for Ubuntu, this will be a viable and efficient approach.  I'm less certain about the homebrew ergonomics.

jamiesnape, 2017-11-01:
Getting Jenkins jobs building the binaries and deploying the artifacts would be a possibility then?

soonho-tri, 2017-11-01:
@jamiesnape , that'd be useful in general. 

For dReal, I think I can provide those artifacts by myself except for the El Capitan case. Honestly, I think we can ignore this combination.

BTW, do we have ETA for CI to have 10.12 and 10.13?

jamiesnape, 2017-11-01:
We already have 10.12. No ETA on 10.13 at present.

soonho-tri, 2017-11-01:
I see. I'll make sure that this mechanism supports macOS 10.13 though we can't test it in CI at the moment.

jamiesnape, 2017-11-02:
Looked again at High Sierra on CI. Apparently there is a command line flag that you can pass to the upgrader from Sierra to High Sierra that prevents the file system upgrade. That seems to have produced images that will boot. Still a fair amount of work to roll out all the images, but should be within a week or two.

soonho-tri, 2017-11-09:
BTW, I got an El Capitan (10.11) VM in the office and generated the bottles for 10.11.

soonho-tri, 2018-01-18:
Close this one since we ended up not going to this direction.

jwnimmer-tri, 2018-01-18:
On Ubuntu, I would still like to stop calling `dpkg` manually to install `dReal` and rather use `deb.bzl` to unpack it within the build tre.  My understanding was that the "install dReal & IBEX as root" was an interim measure.

soonho-tri, 2018-01-18:
I see. I'll leave it as open and work on it (but as a lower priority item).

jamiesnape, 2018-01-18:
We do have a drake apt repo now, from which we could serve the dreal deb, if that helps.

soonho-tri, 2018-01-18:
That's nice. Could you send me how to upload a package there? 

BTW, recently, I've found that launchpad.net does not keep old packages. So I was thinking about moving IBEX's deb (which is currently hosted at launchpad.net) to bintray.com (where homebrew stores its bottles). Moving it to drake apt repo sounds like a good solution (+ dReal).

jamiesnape, 2018-01-18:
At the moment I sign the packages and upload them (we only host libbot and lcm so far), so you would need to send me a links to the debs or the debs themselves. Medium term, there will be a Jenkins job, but I don't have an ETA on that.

soonho-tri, 2018-01-18:
> so you would need to send me a links to the debs 

Here they are:
 - ibex-2.6.3: https://launchpad.net/~dreal/+archive/ubuntu/dreal/+files/libibex-dev_2.6.3.20171215122721.git2275df8f465a9db6a42d497ca322011ff2c6f8f7~16.04_amd64.deb
 - dreal-4.17.12.3: https://dl.bintray.com/dreal/dreal/dreal_4.17.12.3_amd64.deb

jamiesnape, 2018-01-22:
@soonho-tri How difficult would it be to rebuild dreal to install to `/opt/dreal/4.17.12.3`?

jamiesnape, 2018-01-22:
(also IBEX to `/opt/ibex/<version>`, though I am guessing that `.deb` is less likely to change)

soonho-tri, 2018-01-22:
Should be easy. Will do that today.

soonho-tri, 2018-01-23:
@jamiesnape , I have a new working version. Could you check if the following list looks good to you (modulo a specific version number)? Especially, I'd like to check the install location of its pkg-config file.

```
$ dpkg-deb -c dreal_4.18.01.2_amd64.deb | cut -b 49-
./opt/dreal/4.18.01.2/
./opt/dreal/4.18.01.2/share/
./opt/dreal/4.18.01.2/share/doc/
./opt/dreal/4.18.01.2/share/doc/dreal/
./opt/dreal/4.18.01.2/share/doc/dreal/LICENSE.txt
./opt/dreal/4.18.01.2/lib/
./opt/dreal/4.18.01.2/lib/pkgconfig/
./opt/dreal/4.18.01.2/lib/pkgconfig/dreal.pc
./opt/dreal/4.18.01.2/bin/
./opt/dreal/4.18.01.2/bin/dreal
./opt/dreal/4.18.01.2/include/
./opt/dreal/4.18.01.2/include/dreal/
./opt/dreal/4.18.01.2/include/dreal/dreal.h
./opt/dreal/4.18.01.2/include/dreal/version.h
./opt/dreal/4.18.01.2/lib/libdreal.so
./opt/dreal/4.18.01.2/include/dreal/api/
./opt/dreal/4.18.01.2/include/dreal/api/api.h
./opt/dreal/4.18.01.2/include/dreal/smt2/
./opt/dreal/4.18.01.2/include/dreal/smt2/logic.h
./opt/dreal/4.18.01.2/include/dreal/solver/
./opt/dreal/4.18.01.2/include/dreal/solver/config.h
./opt/dreal/4.18.01.2/include/dreal/solver/context.h
./opt/dreal/4.18.01.2/include/dreal/symbolic/
./opt/dreal/4.18.01.2/include/dreal/symbolic/symbolic.h
...
```

> (also IBEX to /opt/libibex/<version>, though I am guessing that .deb is less likely to change)

The current ibex `.deb` file includes paths such as `/usr/include/ibex/...` and `/usr/lib/...`. My understanding is that all the paths should be changed. Is there something I misunderstand?

soonho-tri, 2018-01-23:
>  including the pkg-config file.

I see. Is it our plan to ask users to have `PKG_CONFIG_PATH=/opt/dreal/{version}/lib/pkgconfig/`?

jamiesnape, 2018-01-23:
I think we will need to set it in the bazel build. If there isn't a way to hint to `pkg_config_repository`, we should probably add one. It would help with using keg-only Homebrew packages too.

soonho-tri, 2018-01-23:
Got it. I have made updates for dReal. Will work on ibex later today.

soonho-tri, 2018-01-23:
@jamiesnape , I've uploaded the new `.deb` files at the following locations:
 - https://dl.bintray.com/dreal/dreal/dreal_4.18.01.2_amd64.deb
 - https://dl.bintray.com/dreal/ibex/libibex-dev_2.6.5.20180123154310.gitf618c7b296182f90a84d54936d144b87df0747b9~16.04_amd64.deb

I tried the following to test them:

```bash
# Purge, Checkout the new debs, and install them
sudo dpkg --purge libibex-dev dreal
wget https://dl.bintray.com/dreal/dreal/dreal_4.18.01.2_amd64.deb
wget https://dl.bintray.com/dreal/ibex/libibex-dev_2.6.5.20180123154310.gitf618c7b296182f90a84d54936d144b87df0747b9~16.04_amd64.deb
sudo dpkg --install libibex-dev_2.6.5.20180123154310.gitf618c7b296182f90a84d54936d144b87df0747b9~16.04_amd64.deb
sudo dpkg --install dreal_4.18.01.2_amd64.deb

# Set up PKG_CONFIG_PATH
export PKG_CONFIG_PATH=/opt/dreal/4.18.01.2/lib/pkgconfig:/opt/libibex/2.6.5/share/pkgconfig

# Clean up Drake and build //solvers:dreal_solver_test
bazel clean --expunge
bazel build //solvers:dreal_solver_test
```

All work well up to this point. But the generated `bazel-bin/solvers/dreal_solver_test` has problems.
```bash
$ ldd ./bazel-bin/solvers/dreal_solver_test
...
	libdreal.so => not found
	libibex.so => not found
...
```

jamiesnape, 2018-01-23:
Add `-Wl,-rpath /opt/dreal/4.18.01.2/lib` maybe?

soonho-tri, 2018-01-23:
I guess for now we ignore `-L<path>` part from `pkg-config <package_name> --libs`, which should be passed to our build?

jamiesnape, 2018-01-23:
That would certainly explain the problem.

jamiesnape, 2018-01-23:
What does the BUILD file in `external/dreal` look like?

soonho-tri, 2018-01-23:
Hmm, it *does* have `-L` options in `linkopts`..
```
# -*- python -*-

package(default_visibility = ["//visibility:public"])

cc_library(
    name = "dreal",
    srcs = [],
    hdrs = glob(["include/**"]) + [],
    copts = [],
    defines = [],
    includes = ["include/_opt_dreal_4.18.01.2_include", "include/_opt_libibex_2.6.5_include", "include/_opt_libibex_2.6.5_include_ibex", "include/_opt_libibex_2.6.5_include_ibex_3rd", "include/_usr_include_coin"],
    linkopts = ["-L/opt/dreal/4.18.01.2/lib", "-L/opt/libibex/2.6.5/lib", "-ldreal", "-libex", "-lClp", "-lCoinUtils", "-lbz2", "-lz", "-llapack", "-lblas", "-lm", "-lClp", "-lCoinUtils", "-lbz2", "-lz", "-llapack", "-lblas", "-lm", "-lnlopt", "-lm"],
    deps = [],
)
```

jwnimmer-tri, 2018-01-23:
The pkg_config.bzl logic is a bit incomplete.  As I understand it, we should be creating a `lib` folder with the external workspace, and symlinking the relevant library files into there, as we do with the include paths.  We've ignored that up until now because everything (on Ubuntu at least) was already in `/usr` on the default library load path.

jwnimmer-tri, 2018-01-23:
I guess another option is to rewrite `-L` to `-Wl,-rpath` though I'd want to understand why `-L` isn't working, first.

soonho-tri, 2018-01-23:
Rewriting `-L<path>` to `-L<path> -Wl,-rpath <path>` makes it work. But I have no idea why it works.

jamiesnape, 2018-01-23:
A quick grep of the Bazel source does not find anywhere where they fix up the rpaths, so maybe they just didn't implement that yet.

jamiesnape, 2018-01-23:
They have this line https://github.com/bazelbuild/bazel/blob/ba66e72295d763e43b92179ce6bd46476084a548/src/main/java/com/google/devtools/build/lib/rules/cpp/CppActionConfigs.java#L581 which adds Bazel's known list of paths to the rpath, but I do not see anywhere where they use the link path to set the rpath.

soonho-tri, 2018-06-14:
@jwnimmer-tri , I think we can close this (thanks to https://github.com/RobotLocomotion/drake/pull/8218). What do you think?

jwnimmer-tri, 2018-06-14:
I agree -- its solved for Ubuntu and I think leaving homebrew as-is seems to be working alright.

The one discussion upthread that might still be relevant is that we should be sure that Drake maintainers besides @soonho-tri can create and distribute new binary releases.  If there's not already an issue tracking that, perhaps there should be.

jwnimmer-tri, 2018-06-14:
I guess https://github.com/dreal/dreal4#build-debian-package is a fair start, but some kind of CI-related automation would be super.

closed 2018-06-14 by jwnimmer-tri

soonho-tri, 2018-06-14:
FYI,

1. Correct, the `#build-debian-package` page includes the instruction (one-liner).
2. The latest and the previous versions of `.deb` files are also available at https://bintray.com/dreal/dreal/dreal#files.