RobotLocomotion/drake #7399: Documentation incomplete on PiecewisePolynomial: why is the coefficient matrix a matrix and not a vector?

edrumwri, 2017-11-02:
A reasonable examination of both the doxygen and the source code for piecewise polynomial did not reveal what the rows and columns in the coefficient matrix correspond to. An example that I am using indeed uses only row vectors. Is using a matrix only sugar, as appears to be the case?

edrumwri, 2017-11-02:
Siyuan, please feel free to reassign, as appropriate. You appear to be original author of this file?

siyuanfeng-tri, 2017-11-02:
This legacy code from drc era. I just happened to be the one slightly updating it. I didn't find it odd, it's rather convenient if you want to have a spline for each element of a N by M thing.

edrumwri, 2017-11-03:
Ah, then it is sugar. It's fine as a design decision, but I spent a good fifteen minutes trying to figure out what needed to go in the rows (in other words, some documentation would have been very much appreciated!)

jwnimmer-tri, 2019-03-04:
@edrumwri Maybe you can PR some suggested documentation that would have preempted your confusion?

PR #10844 (2019-03-07): Updates the documentation for PiecewisePolynomial.
Resolves #7399

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10844)
closed 2019-03-07 by edrumwri