RobotLocomotion/drake #7412: Resolving system dependency conflicts (meta)

stonier, 2017-11-06:
**FCL & Dependencies**

* [ ] **ccd** : send patch upstream for symbol hiding
* [ ] **ccd** : patch our fork and statically build
* [ ] **octomap** : send patch upstream for symbol hiding
* [ ] **octomap** : patch our fork and statically build
* [x] **fcl** : send patch upstream for symbol hiding ([fcl/pull/233](https://github.com/flexible-collision-library/fcl/pull/233))
* [ ] **fcl** : cherry-pick the patch that allows symbols to be hidden and statically build

**YAML**

* [x] Drop back to system yaml, (~~[drake/pull/7352](https://github.com/RobotLocomotion/drake/pull/7352)~~)

**VTK**

* [ ] Privatise and do not install vtk_utils.h, ([drake/pull/7258](https://github.com/RobotLocomotion/drake/pull/7258/files))

**PCL Enabling**

* [ ] **eigen** : workaround PCLConfig.cmake issues with Eigen, ([shambhala#38](https://github.com/RobotLocomotion/drake-shambhala/issues/38))
* [ ] **vtk** : resolve a vtk linking bug that occurs when compiling against drake & pcl

**Protobuf**

* [x] ~~initial idea and working notes ([drake/pull/7278](https://github.com/RobotLocomotion/drake/pull/7278))~~
* [ ] fall back to system protobuf 2.6 ([drake/pull/7323](https://github.com/RobotLocomotion/drake/pull/7323))
* [ ] decouple drake from protobuf (optional)

More detailed notes in this [spreadsheet](https://docs.google.com/spreadsheets/d/1tiSN8LtNCgos-Vt3U4IBt8okEIC5CKiYLdLx0yo9EfU/edit#gid=863450394).

@nuclearsandwich can you add comments as things get resolved, or new items are brought to the fore. I'll update this list as we go.

jwnimmer-tri, 2017-11-06:
FYI I reduced the permissions on the "spreadsheet" link above from "Anyone with the link can edit" to "Anyone with the link can view".  Otherwise, the document would turn into a w4r3z site.  We may have to add back individually-authenticated commentors or editors later to compensate.

stonier, 2017-11-07:
From the `WORKSPACE` file, it looks as if we are actually not currently using forks of `fcl` nor `libccd`, just hashes from the original projects. They both certainly are getting upstream patches.

**libccd**

We are in fact using a very old tag of libccd (2014). There has been very little code changes since then - just one commit I think. The rest have been in CMake upgrade and it looks like @jamiesnape has already started to do some symbol hiding. Diff between tip and v2.0 [here](https://github.com/danfis/libccd/compare/v2.0...master). Since it already has some logic for exports in tip, we should probably try and upgrade Drake to use that and do whatever patching is necessary on top of that to hide the symbols in Drake.

* @j-rivero : check that Drake builds from tip
* @j-rivero : if that works, create a PR that points to tip of the [RobotLocomotion/libccd](https://github.com/RobotLocomotion/libccd) fork I just created
* @avalenzu : to follow up and make sure that the contact collisions still work with the PR
* @j-rivero : if necessary, drop a patch on the fork for hiding symbols, update the PR, and upstream the patch

Sound like a plan?

**FCL**

It has a fork in [RobotLocomotion/fcl](https://github.com/RobotLocomotion/fcl) now.

j-rivero, 2017-11-07:
> Sound like a plan?

Yep. Working on it.

j-rivero, 2017-11-07:
@stonier the pull request should be ready at pull request #7417 

The fcl cherry pick for your own fork is at https://github.com/RobotLocomotion/fcl/pull/1

stonier, 2017-11-09:
Some updates to the above.

**Yaml** 

Yaml is blocking on system protobuf getting in because of a Mac issue on crosstool not having `/usr/local/include` which indirectly itself causes issues with the protobuf 3.1 currently being used. Refer to the causal chain in the first post.

**Protobuf**

Pending on a solution to adopting homebrew's python. @jamiesnape can you raise that issue and start discussion on how to solve this ASAP?

jamiesnape, 2017-11-09:
> Pending on a solution to adopting homebrew's python. @jamiesnape can you raise that issue and start discussion on how to solve this ASAP?

#7440

jamiesnape, 2017-11-09:
> We are in fact using a very old tag of libccd (2014). There has been very little code changes since then - just one commit I think. The rest have been in CMake upgrade and it looks like @jamiesnape [...]

You should be able to use all the logic I added to hide everything that you need to and keep their three build systems working rather than trying to do so with CMake-specific logic per https://github.com/danfis/libccd/pull/39. Not using `GenerateExportHeader` was deliberate given the requirements of that project, though maybe that changed.

nuclearsandwich, 2017-11-09:
> - [ ] eigen : workaround PCLConfig.cmake issues with Eigen, (shambhala#38) https://github.com/RobotLocomotion/drake/pull/7435
> - [x] vtk : resolve a vtk linking bug that occurs when compiling against drake & pcl https://github.com/RobotLocomotion/drake/pull/7434

nuclearsandwich, 2017-11-10:
https://github.com/RobotLocomotion/drake/pull/7453 excludes the yaml_util header from the installation without reverting back to system yaml-cpp. We probably still want to re-introduce the system yaml-cpp changes once the build on mac stabilizes.

j-rivero, 2017-11-10:
> Not using GenerateExportHeader was deliberate given the requirements of that project, though maybe that changed.

To pass all the CI tests (they are using cmake, autotools and Makefile) you are right, we can not rely only in the `GenerateExportHeaders` of cmake. I've changed the PR to use an static file for the exports and patch Makefile and autotools. All tests passes now.

jamiesnape, 2017-11-10:
To cut down the size of the diff can you keep the macros named `_ccd_export` and keep them in `compiler.h`. Also you can keep the `project(libccd C)` line now, you have a reference to FCL instead of CCD in one of the comments, and their CMake style is to have a continued line indent of two and no space before opening parentheses in the files.

jamiesnape, 2017-11-14:
Looks like libccd merged the patch.

jwnimmer-tri, 2018-01-17:
I think this might be finished?  At least, the checkbox list in the top post shows that everything has merged.

nuclearsandwich, 2018-01-19:
Every dependency with an identified conflict was resolved. So yes, unless/until new issues are reported I think this could be closed.

jwnimmer-tri, 2018-01-19:
Thanks!  This was very exciting to see happen, it should be a nice improvement in compatibility.  I will close for now but we can re-open and/or file a new issue if we find we need other improvements.

closed 2018-01-19 by jwnimmer-tri