RobotLocomotion/drake #7420: Add support for integer- and/or binary-valued states and parameters

jadecastro, 2017-11-08:
This would be useful for writing Mixed-Integer problems over System ContinuousStates and Parameters in a natural way, much like we do for BasicVectors.  For instance, we could then write Systems and Diagrams with named vectors deriving from a vector of discretes rather than doing so from only BasicVectors.

The larger picture is that we could then use these phrasings to have discrete analogues for MathematicalPrograms, for instance mixed-integer trajectory optimization.

sherm1, 2017-11-08:
Are the current abstract states and parameters inadequate for this purpose? Can you clarify where they fall short?

RussTedrake, 2017-11-14:
fwiw -- i understand the request (thought about it before myself), but wasn't able to decide for myself what the right solution is (yet).  Might be a good group discussion at some point.

@sherm1 -- the advantage of doing more than abstract state is that methods outside of the class could use the systems framework to set up more structured formulations.  
@jadecastro -- on the other hand, it's possible we could accomplish this by having an AbstractValue (or even BasicVector) state which is only well defined at integer values, and then register a system constraint declaring the domain of those variables to be integers.

RussTedrake, 2017-11-14:
Do we have a good example yet (or any example) of actually doing symbolic or even autodiff through an AbstractValue?

+ @soonho-tri

sherm1, 2021-11-11:
AbstractState provides this capability already.

closed 2021-11-11 by sherm1