RobotLocomotion/drake #7451: <gtest/gtest.h> is used in public headers

jamiesnape, 2017-11-10:
Do not think we have a tracking issue, but seeing `<gtest/gtest.h>` in used in the install tree was slightly surprising.

For completeness, this is the current list of external dependencies used in public headers are:
```
<bot_core/atlas_command_t.hpp>
<bot_core/pointcloud_t.hpp>
<bot_core/position_3d_t.hpp>
<bot_core/quaternion_t.hpp>
<bot_core/twist_t.hpp>
<bot_core/vector_3d_t.hpp>
<btBulletCollisionCommon.h>
<BulletCollision/NarrowPhaseCollision/btGjkEpaPenetrationDepthSolver.h>
<BulletCollision/NarrowPhaseCollision/btGjkPairDetector.h>
<BulletCollision/NarrowPhaseCollision/btPointCollector.h>
<Eigen/Core>
<Eigen/Dense>
<Eigen/Eigenvalues>
<Eigen/Geometry>
<Eigen/LU>
<Eigen/SparseCore>
<Eigen/StdVector>
<fcl/fcl.h>
<gflags/gflags.h>
<google/protobuf/arena.h>
<google/protobuf/arenastring.h>
<google/protobuf/extension_set.h>
<google/protobuf/generated_enum_reflection.h>
<google/protobuf/generated_message_util.h>
<google/protobuf/io/zero_copy_stream_impl.h>
<google/protobuf/message.h>
<google/protobuf/metadata.h>
<google/protobuf/repeated_field.h>
<google/protobuf/stubs/common.h>
<google/protobuf/unknown_field_set.h>
<gtest/gtest.h>
<ignition/math/Spline.hh>
<ignition/math/Vector3.hh>
<ignition/rndf/UniqueId.hh>
<lcm/lcm_coretypes.h>
<lcm/lcm-cpp.hpp>
<lcmtypes/bot_core/robot_state_t.hpp>
<optitrack/optitrack_frame_t.hpp>
<robotlocomotion/image_array_t.hpp>
<spdlog/fmt/ostr.h>
<spdlog/spdlog.h>
<stx/optional.hpp>
<tinyxml2.h>
<unsupported/Eigen/AutoDiff>
<unsupported/Eigen/Polynomials>
<yaml-cpp/yaml.h>
```

Aside: It feels like the use of `spdlog`, `tinyxml2`, `yaml-cpp` could probably be refactored or headers made private to not appear in this list. I know that work is ongoing to remove `fcl` (which would cover transitive usage of `libccd` and `octomap` that I have not included here.)

jwnimmer-tri, 2017-11-10:
This is great visibility, thanks.  I'm thinking we should have more tests like [drake/bindings/python/pydrake/test/testCommonInstall.py](https://github.com/RobotLocomotion/drake/blob/master/drake/bindings/python/pydrake/test/testCommonInstall.py) that acceptance test the installation.  Sanity-checking the list of non-Drake includes in Drake headers would be a great one.

jamiesnape, 2017-11-10:
#7455 adds `<sdf/sdf.hh>`.

amcastro-tri, 2017-11-10:
I don't know if it matters, but `<sdf/sdf.hh>` was introduced in #6496

jamiesnape, 2017-11-10:
That is just in an example, so not in a public header.

jamiesnape, 2018-11-06:
Apparently VTK found its way into public headers, or more precisely I think #9830 made public some headers that include VTK.

jwnimmer-tri, 2020-10-16:
Here's a brief update as of v0.23.0:

```
# Unwanted includes (i.e., recent regressions).
<ignition/...>
<fcl/...>

# Built from source.
<fmt/...>
<lcm/...>
<spdlog/...>

# LCM message types.
"bot_core/..."
"optitrack/..."
"robotlocomotion/..."

# Coming from Ubuntu or macOS host system:
<Eigen/...>
<tinyxml2.h>
<yaml-cpp/...>
```

jwnimmer-tri, 2021-07-24:
Update as of v0.33.0:

```
# Built from source.
#include <lcm/...>

# LCM message types.
#include <optitrack/...>

# Coming from Ubuntu or macOS host system:
#include <Eigen/...>
#include <fmt/...>     // (built from source on Bionic)
#include <spdlog/...>  // (built from source on Bionic)
#include <tinyxml2.h>
#include <yaml-cpp/yaml.h>
```

jwnimmer-tri, 2021-09-24:
Update as of v0.34.0:

```
# Built from source.
#include <lcm/...>

# LCM message types.
#include <optitrack/...>

# Coming from Ubuntu or macOS host system:
#include <Eigen/...>
#include <fmt/...>     // (built from source on Bionic)
#include <spdlog/...>  // (built from source on Bionic)
#include <yaml-cpp/yaml.h>
```

The only change that `tinyxml2.h` is no longer in public headers.

jwnimmer-tri, 2023-06-14:
Update as of v1.17.0:

```
# Built from source.
#include <lcm/...>

# LCM message types.
#include <optitrack/...>

# Coming from Ubuntu or macOS host system:
#include <Eigen/...>
#include <fmt/...>
#include <spdlog/...>
```

See #19600 for the next planned purge (optitrack messages).

closed 2024-03-16 by jwnimmer-tri