RobotLocomotion/drake #7467: g++<=4.9's does not recognize clang's [abi:c++11] tags, produces linker errors in drake_cmake_installed-like repos

EricCousineau-TRI, 2017-11-14:
When @naveenoid was building a `catkin`-based downstream project for Drake, he had run into a linker error when trying to build his library in CMake:
```
undefined reference to `RigidBodyTree<double>::get_position_name(int) const'
```
However, `get_position_name(...)` was clearly defined, and other binaries (built in Bazel) were able to find this symbol.

Upon investigation, it was because Drake was being compiled with `clang`, while his project was compiled with `g++` (the default compiler for his system). Digging slightly deeper revealed that this ABI compatibility incompatibility only seems to happen at or before `g++-4.9`.
As always, StackOverflow had an answer: https://stackoverflow.com/q/36159238/7829525

See [repro](https://github.com/EricCousineau-TRI/repro/tree/bd6e9ea1c05bbd0067ad5d07e1679cf9ed84cab9/bug/gcc_clang_abi) and [example output](https://github.com/EricCousineau-TRI/repro/blob/bd6e9ea1c05bbd0067ad5d07e1679cf9ed84cab9/bug/gcc_clang_abi/repro.output.txt).

Possible courses of action:

* In the generated `drake-config.cmake`, have it check the CXX compiler; if it detects `gcc<=4.9` (and if it was generated with `clang`), stop and puke errors.
* In some headers in `//drake/common`, add a version check to prevent any compilation for `gcc<=4.9`.
* Add a check in `install_prereqs.sh` (something that may be similar to what we may do for the SIP `/usr/bin/python` on Mac?) to flag this as an issue.

At the time, the workaround we chose was to simply `update-alternatives` to switch `cc` / `c++` from `gcc` / `g++` to `clang` / `clang++`.

jwnimmer-tri, 2017-11-14:
FYI Never use update-alternatives to change the compiler your project it using.  See for example https://askubuntu.com/questions/26498/choose-gcc-and-g-version/26500#26500 or tangentially a#2830.  Typically, the solution is instead to set `CC` and/or `CXX` variables to point to the compiler you want.

jamiesnape, 2017-11-15:
Neither `g++-4.9` or Catkin are actually supported by Drake, of course. Catkin support will appear in the future, but `g++-4.9` support is long gone.

EricCousineau-TRI, 2017-11-15:
Understood, but could we possibly throw an error in CMake if a user tries to consume Drake using `g++-4.9`, so that they have a (much much) more informative error?

jamiesnape, 2017-11-15:
I am inclined to say no since CMake supports so many compilers. The usual way to do these sorts of things is to specify [compile features](https://cmake.org/cmake/help/v3.5/manual/cmake-compile-features.7.html#manual:cmake-compile-features(7)).

jamiesnape, 2017-11-15:
Looks like we currently only require `cxx_decltype` and `cxx_decltype_auto` from [this list](https://cmake.org/cmake/help/v3.5/prop_gbl/CMAKE_CXX_KNOWN_FEATURES.html#prop_gbl:CMAKE_CXX_KNOWN_FEATURES)

EricCousineau-TRI, 2017-11-15:
Gotcha. In that case, I will go ahead and update the FAQ with this failure mode.
Thanks!

jamiesnape, 2017-11-15:
An open question is whether packages should be generated with Clang 4.0, GCC 5.4, or both. CMake certainly tends to favor GCC as its default on most platforms for various (sometimes indirect) reasons.

jwnimmer-tri, 2017-11-15:
FYI I would be OK with GCC 5.4 for Xenial binary releases; its the one used by `build-essential`.  (I like Clang 4.0 as the default for from-source builds, as its error messages still seem to be better than GCC.)

EricCousineau-TRI, 2017-11-21:
Closing via #7491.

closed 2017-11-21 by EricCousineau-TRI