RobotLocomotion/drake #7482: Consider removing MATLAB mathematical_program_test

jamiesnape, 2017-11-15:
Basically the calling of C++ through Python through MATLAB is proving more and more troublesome and harder and harder to debug as we add more system dependencies, such as in #7323. It also adds an extra way that #7440 needs to be fixed and limits our ability to isolate conflicting dependencies from MATLAB if they are also used by Python. I would like to propose that we remove the test completely and concentrate efforts on the either the Protobuf way of interfacing with MATLAB or some pure C++ way of doing so.

FYI @stonier.

EricCousineau-TRI, 2017-11-15:
Related: #5981 

Just to check, would the Protobuf way be the most robust? (but, being IPC, possibly cut off certain workflows, or at least require additional re-engineering of the existing interfaces?)
Do we know what workflows want to support in MATLAB?

Going pure C++ seems like it would only sidestep #7440, which seems like a short-term decision could be made with relatively low implementation barriers? (Option 2 possibly being the easiest short-term, but less contained than say Option 3 or 5 in that issue, via `install_prereqs.sh`?)

jamiesnape, 2017-11-15:
> Just to check, would the Protobuf way be the most robust?

It would pretty much air gap MATLAB and its internal dependencies.

> Going pure C++ seems like it would only sidestep #7440, which seems like a short-term decision could be made with relatively low implementation barriers?

#7440 is a very minor concern as it related to this issue. The main issue is the practicalities and amount of resources it takes to support this one test.

jamiesnape, 2017-11-15:
> (Option 2 possibly being the easiest short-term, but less contained than say Option 3 or 5 in that issue, via install_prereqs.sh?)

Note that option 2 does not fix MATLAB.

jwnimmer-tri, 2017-11-15:
Without yet stating an opinion with respect to direct-use linking vs IPC, I want to clarify that the purpose of `mathematical_program_test` was to be the canary that proves that the MATLAB -> Python -> Drake direct-use paradigm keeps working, since that paradigm was the plan of record for how Drake continues to be available to MATLAB users.  (The idea was to expand and add capture more APIs with similar tests, once the build system transition completed.)  If we want to change our MATLAB user roadmap to a different paradigm we can do that, but we should not disable the test merely because it is causing headaches; we should only disable it if we decide to stop supporting that paradigm for the foreseeable future.

jamiesnape, 2017-11-15:
Yes, I am basically saying that paradigm is and will be a support nightmare, and we should look elsewhere for a solution.

jamiesnape, 2018-01-04:
Closing since there seems little support beyond myself for the proposal.

jwnimmer-tri, 2018-01-18:
For the record (I'm finally catching up on some email), I think its well put that trying to load `libdrake.so` into MATLAB's memory space is going to be an ongoing challenge, as it stands now.  To me, its likely that either we will need to use a IPC mechanism "air gap" as alluded to above, or else to link a special flavor of `libdrake.so` for use by MATLAB, where only the python binding symbols are public and everything else (other than libc / libstdc++) are hidden visibility and statically linked.  (This might also mean we have a monolithic bindings so, instead of the several that we use now.)

If we don't have an open issue to work toward some kind of de-coupling or remoting, maybe we should?  (Or reopen this one with a new title?)  While I'm not sure about turning off the test in the meantime, it does seem worthwhile to document the pain point and work on resolving it.

RussTedrake, 2018-01-18:
Sounds reasonable to me.  I think we are learning a lot about the python bindings right now, and will have a much more well-informed opinion about this within even a few more weeks.

PR #8215 (2018-03-01): Remove MATLAB mathematical_program_test and dead MATLAB code
Closes #7482. FYI @RussTedrake @jwnimmer-tri.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8215)
commit 74097bdf (2018-03-01): Remove MATLAB mathematical_program_test and dead MATLAB code

Closes #7482.
closed 2018-03-01 by jamiesnape