RobotLocomotion/drake #7484: Support both gurobi 7.0.2 and 7.5.2

hongkai-dai, 2017-11-15:
Our CI uses gurobi 7.0.2, but gurobi has a new version 7.5.2 that is faster. Our bazel build code hardcodes gurobi 7.0.2 in https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/gurobi/gurobi.bzl#L15

hongkai-dai, 2017-11-15:
It would be better if we could support both gurobi 7.0.2 and 7.5.2 in the bazel build

jwnimmer-tri, 2017-11-15:
@hongkai-dai Would there be a problem if we upgraded CI to use 7.5.2 and made that version required?  (In other words, why should we support 7.0.2 at all?)

hongkai-dai, 2017-11-15:
Your are right, it is better to support gurobi 7.5.2 only. I know there are some students in Russ's group using gurobi, I do not know which version do they use.

@gizatt will it be OK if we upgrade to gurobi 7.5.2 for your use case?

jamiesnape, 2017-11-15:
Looks like the same license is good for both 7.0.2 and 7.5.2 and the 7.5.2 packages are already downloaded to the Xenial images. We would just need to install 7.5.2 onto the Mac images and work out the logistics of the switchover.

gizatt, 2017-11-15:
Yup, fine with me!

jamiesnape, 2017-11-15:
@BetsyMcPhail we will need to install 7.5.2 onto the Mac images, but not entirely sure the two versions can coexist. Can you check whether they will and, if so, run the installer on each of the images? You can grab it from the `drake-provisioning` bucket on S3.

jamiesnape, 2017-11-16:
Do they actively conflict, though? Looks like Gurobi 7.0.2 is in a versioned folder in `/Library`.

BetsyMcPhail, 2017-11-16:
Sorry, I misunderstood the question.

Multiple versions of Gurobi can happily coexist. Gurobi 7.5.2 has been installed (/Library/gurobi752) on both the Sierra and High Sierra images and Jenkins has been updated.

PR #7540 (2017-12-01): Update supported Gurobi version to 7.5.2
Closes #7484.

This will fail Linux CI initially since GUROBI_PATH is still set to 7.0.2.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7540)
closed 2017-12-01 by jamiesnape