RobotLocomotion/drake #7487: Add support for a `call_matlab`-like interface for Python

EricCousineau-TRI, 2017-11-15:
Per @RussTedrake's request, also enabling road warriors who may not have Internet access for MATLAB (@avalenzu).

I have a prototype here: [branch](https://github.com/EricCousineau-TRI/drake/blob/feature/call_py/drake/common/proto/test/call_python_test.cc) | [commit](https://github.com/EricCousineau-TRI/drake/blob/c62de4e/drake/common/proto/test/call_python_test.cc)
(Still figuring out issues with column-major storage and NumPy, so 3d plot is still off.)

Proposed PR train:

- [ ] Add initial Python functionality on top of `call_matlab` code.
- [ ] Add multi-threading code setup.
- [ ] Reshuffle code to have the core RPC stuff be non-MATLAB specific.

RussTedrake, 2017-11-16:
love it, love it, love it.  (especially the "print", "Howdy").  
the initial python functionality is all i need for now.  the second two proposed PRs are great, but definitely should not block the first.  
Thank you!

EricCousineau-TRI, 2017-11-27:
Per @jamiesnape's [comment](https://github.com/RobotLocomotion/drake/pull/7496#issuecomment-347209543), I've added the 4th bullet point.
My inclination is that installation would be a little more scoped if this client were to be part of `pydrake` for now. If, for whatever reason, someone wants to use this code outside of Drake, we can split it out at that point.

EricCousineau-TRI, 2017-12-06:
Per @hongkai-dai's question, I've inserted the new 3rd bullet point (for documenting it).

PR #10168 (2018-12-10): call_python: Add documentation example
Resolves #7487 (addressing documentation)

Figgered this was simple enough to close out.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10168)
closed 2018-12-10 by sammy-tri