RobotLocomotion/drake #7505: MATLAB Documentation in `Binary Installation`is out-of-date.

EricCousineau-TRI, 2017-11-17:
In revising some text for #7467, it looks like the current MATLAB-related docs for binary installation is out of date:
http://drake.mit.edu/from_binary.html#running-matlab-examples
https://github.com/RobotLocomotion/drake/blob/758e7051c3fc42ea22960919ab68398f40694806/doc/from_binary.rst#running-matlab-examples

Should we just remove it since we already have documentation regarding the MATLAB-Python unittests elsewhere?

jwnimmer-tri, 2017-11-17:
That sub-section is part of "Using older (2015) releases" section.  I think its still accurate (not out-of-date) as far as it goes.  I guess it depends on whether we want to continue to document how to use the past examples, and if so perhaps cleaning up the doc organization to make things clearer.

EricCousineau-TRI, 2017-11-17:
D'oh, did not look at the font size difference. Will close for now, considering clean up at a future point.

closed 2017-11-17 by EricCousineau-TRI