RobotLocomotion/drake #7507: python crashes when trying to import pydrake on Mac

pangtao22, 2017-11-21:
After running `bazel run install -- ~/drake-build` and add `~/drake-build/lib/python2.7/site-packages` to `PYTHONPATH`, python interpreter crashed when I tried to import pydrake on Mac:
```
dhcp-18-189-90-216:drake-distro pangtao$ python
Python 2.7.14 |Anaconda, Inc.| (default, Oct  5 2017, 02:28:52)
[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pydrake
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
dhcp-18-189-90-216:drake-distro pangtao$
```

jamiesnape, 2017-11-22:
We only support the Homebrew Python at present. Anaconda is not supported.

pangtao22, 2017-11-23:
Tried home-brew python and had the same problem:
```
30-9-79:drake-distro pangtao$ python
Python 2.7.10 (default, Jul 15 2017, 17:16:57)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pydrake
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
30-9-79:drake-distro pangtao$
```

jamiesnape, 2017-11-23:
That is the system Python. You need to either use the `python2` executable or set `export PATH="/usr/local/opt/python/libexec/bin:$PATH"`. See `brew info python` for more information.

pangtao22, 2017-11-28:
Thanks. Pydrake works perfectly after setting `PATH` to where the homebrew python executable is. 
It would be great if you could help update the pydrake instructions on Drake's website. The current version is very out of date. I had to figure out how to setup pydrake by a mixture of asking around, searching the DrakeDevelopers slack channel and tinkering. 

Below are the steps I followed to setup pydrake on both Mac and Ubuntu. The steps are not platform-specific unless otherwise stated. I hope they will be helpful for updating the instructions. 

1. Clone drake repo into a local folder, setup remotes, and run the pre-build setup script by following the instructions in http://drake.mit.edu/from_source.html. 

2. run `bazel run //:install -- /path/to/drake/install/folder/`
2-1. If using Gurobi, follow Drake's instruction on Gurobi setup, and run `bazel run --config=gurobi //:install -- /path/to/drake/install/folder/`

3. Include pydrake in `PYTHONPATH` by adding  `export PYTHONPATH=$PYTHONPATH:/path/to/drake/install/folder/lib/python2.7/site-packages` to `.bashrc` or `.bash_profile`.

4. Make sure that homebrew python is used:
`export PATH="/usr/local/opt/python/libexec/bin:$PATH"`
p.s. pydrake seems to work just fine with anaconda python on Ubuntu, but crashes on Mac. 

5. On ubuntu, it may be necessary to include the following line in `.bashrc`
`LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/drake/install/folder/lib/python2.7/site-packages`

EricCousineau-TRI, 2018-01-28:
Closed by #7598.

RussTedrake, 2018-01-28:
the last comment by jamie is still super important and easy for people to hit on mac.  can we add it to the docs, too?
> You need to either use the python2 executable or set export 
> PATH="/usr/local/opt/python/libexec/bin:$PATH". See brew info python for more information.

EricCousineau-TRI, 2018-01-29:
Oops, yup! Just submitted #7894.

PR #7894 (2018-01-30): doc/python_bindings: Add caveat for python2 on Mac.
This addresses Russ's [follow-up comment](https://github.com/RobotLocomotion/drake/issues/7507#issuecomment-361097496) in #7507.

Closes #7507.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7894)
closed 2018-01-30 by EricCousineau-TRI