RobotLocomotion/drake #7513: build failure

BobLiao1987, 2017-11-22:
Hi, I have tried several times to build drake, ubuntu16.04, 64bit, however i encounter below:

bob-ros@bobros-Think:~/drake/drake-distro$ bazel build //... 
WARNING: ignoring http_proxy in environment.
ERROR: build interrupted
INFO: Elapsed time: 142.198s
FAILED: Build did NOT complete successfully (0 packages loaded)

bob-ros@bobros-Think:~/drake/drake-distro$ bazel test //...  
WARNING: ignoring http_proxy in environment.
ERROR: build interrupted
INFO: Elapsed time: 306.610s
FAILED: Build did NOT complete successfully (0 packages loaded)
ERROR: Couldn't start the build. Unable to run tests

bob-ros@bobros-Think:~/drake/drake-distro$ bazel build --compiler=gcc-5 //...  
WARNING: ignoring http_proxy in environment.
ERROR: /home/bob-ros/drake/drake-distro/drake/multibody/collision/BUILD.bazel:37:1: no such package '@bullet//': Error downloading [https://github.com/bulletphysics/bullet3/archive/2.86.1.tar.gz, https://drake-mirror.csail.mit.edu/github/bulletphysics/bullet3/2.86.1.tar.gz, https://s3.amazonaws.com/drake-mirror/github/bulletphysics/bullet3/2.86.1.tar.gz] to /home/bob-ros/.cache/bazel/_bazel_bob-ros/aeb2cce4621bd070fcf8d1a0ba4ce510/external/bullet/2.86.1.tar.gz: Checksum was a52c1eb791618af9ade3678e17cb0b463f65adf2349c9f1fe901fb3bc5e35ccb but wanted c058b2e4321ba6adaa656976c1a138c07b18fc03b29f5b82880d5d8228fbf059 and referenced by '//drake/multibody/collision:bullet_collision'
ERROR: Analysis of target '//drake/multibody/collision:collision_filter_group_test_dsym' failed; build aborted
INFO: Elapsed time: 391.187s
FAILED: Build did NOT complete successfully (0 packages loaded)

jamiesnape, 2017-11-22:
Are you behind a proxy? What proxy related environment variables do you have set?

BobLiao1987, 2017-11-23:
yes, I am behind a proxy of shadowsocks. If no proxy, I still get these errors above because of GFW blocking.
I have no idea about the  environment variables that proxy related. Do you have some tips about how to solve this?@jamiesnape

jamiesnape, 2017-11-23:
Not sure about proxies and Bazel. There was this issue, which may help: https://github.com/bazelbuild/bazel/issues/587

Otherwise, try removing these lines to exclusively use our mirrors instead of GitHub and Bitbucket:
* https://github.com/RobotLocomotion/drake/blob/390157bfe269a082727c9cb43bd2ec74cc42e37f/tools/workspace/bitbucket.bzl#L45-L46
* https://github.com/RobotLocomotion/drake/blob/390157bfe269a082727c9cb43bd2ec74cc42e37f/tools/workspace/github.bzl#L43-L44

If you are still blocked by GFW, let me know.

BobLiao1987, 2017-11-25:
Thank you  for you reply, the below is what I have got :
the failure is due to protobuf checksum always change.
so I made these change 
`

--- a/WORKSPACE
+++ b/WORKSPACE
@@ -338,8 +338,8 @@
 github_archive(
     name = "protobuf",
     repository = "google/protobuf",
-    commit = "v3.1.0",
-    sha256 = "fb2a314f4be897491bb2446697be693d489af645cb0e165a85e7e64e07eb134d",  # noqa
+    commit = "v3.2.0",
+    sha256 = "a839d3f1519ff9d68ab908de5a0f269650ef1fc501c10f6eefd4cae51d29b86f",  # noqa
 )
 
 pypi_archive(
diff --git a/tools/workspace/bitbucket.bzl b/tools/workspace/bitbucket.bzl
index 7691b5b..ae052af 100644
--- a/tools/workspace/bitbucket.bzl
+++ b/tools/workspace/bitbucket.bzl
@@ -43,8 +43,8 @@ def bitbucket_archive(
     # bucket.
     mirrors = [
         "https://bitbucket.org/%s/get/%s.tar.gz",
-        "https://drake-mirror.csail.mit.edu/bitbucket/%s/%s.tar.gz",
-        "https://s3.amazonaws.com/drake-mirror/bitbucket/%s/%s.tar.gz",
+        #"https://drake-mirror.csail.mit.edu/bitbucket/%s/%s.tar.gz",
+        #"https://s3.amazonaws.com/drake-mirror/bitbucket/%s/%s.tar.gz",
     ]
     urls = [mirror % (repository, commit) for mirror in mirrors]
 
diff --git a/tools/workspace/github.bzl b/tools/workspace/github.bzl
index bbe8403..6a8aeb0 100644
--- a/tools/workspace/github.bzl
+++ b/tools/workspace/github.bzl
@@ -41,8 +41,8 @@ def github_archive(
     # Packages are mirrored from GitHub to CloudFront backed by an S3 bucket.
     mirrors = [
         "https://github.com/%s/archive/%s.tar.gz",
-        "https://drake-mirror.csail.mit.edu/github/%s/%s.tar.gz",
-        "https://s3.amazonaws.com/drake-mirror/github/%s/%s.tar.gz",
+    #    "https://drake-mirror.csail.mit.edu/github/%s/%s.tar.gz",
+    #    "https://s3.amazonaws.com/drake-mirror/github/%s/%s.tar.gz",
     ]

`

bob-ros@bobros-Think:~/drake/drake-distro$ bazel build //... 
ERROR: build interrupted
INFO: Elapsed time: 28.644s
FAILED: Build did NOT complete successfully (0 packages loaded)
    currently loading: tools/vector_gen ... (5 packages)
bob-ros@bobros-Think:~/drake/drake-distro$ bazel test //...  
INFO: Analysed 4565 targets (101 packages loaded).
INFO: Found 1850 targets and 2715 test targets...
INFO: From Executing genrule //drake/automotive/models:speed_bump_genrule:
[2017-11-25 13:43:31.462] [console] [info] Loading road geometry.
[2017-11-25 13:43:31.463] [console] [info] Generating OBJ.
FAIL: //tools/workspace:bazel_codestyle (see /home/bob-ros/.cache/bazel/_bazel_bob-ros/aeb2cce4621bd070fcf8d1a0ba4ce510/execroot/drake/bazel-out/clang-4.0-linux-opt/testlogs/tools/workspace/bazel_codestyle/test.log)
INFO: From Testing //tools/workspace:bazel_codestyle:
==================== Test output for //tools/workspace:bazel_codestyle:
tools/workspace/github.bzl:44:5: E122 continuation line missing indentation or outdented
tools/workspace/github.bzl:45:5: E122 continuation line missing indentation or outdented
================================================================================
INFO: Elapsed time: 4678.326s, Critical Path: 220.92s
INFO: Build completed, 1 test FAILED, 14763 total actions
//tools/workspace:bazel_codestyle                                        FAILED in 0.2s
  /home/bob-ros/.cache/bazel/_bazel_bob-ros/aeb2cce4621bd070fcf8d1a0ba4ce510/execroot/drake/bazel-out/clang-4.0-linux-opt/testlogs/tools/workspace/bazel_codestyle/test.log

Executed 2715 out of 2715 tests: 2714 tests pass and 1 fails locally.
There were tests whose specified size is too big. Use the --test_verbose_timeout_warnings command line option to see which ones these are.
bob-ros@bobros-Think:~/drake/drake-distro$ bazel build --compiler=gcc-5 //... 
INFO: Analysed 4565 targets (1 packages loaded).
INFO: Found 4565 targets...
INFO: From Executing genrule //drake/automotive/models:speed_bump_genrule:
[2017-11-25 16:13:23.451] [console] [info] Loading road geometry.
[2017-11-25 16:13:23.452] [console] [info] Generating OBJ.
INFO: Elapsed time: 4199.711s, Critical Path: 124.29s
INFO: Build completed successfully, 12013 total actions
bob-ros@bobros-Think:~/drake/drake-distro$

BobLiao1987, 2017-11-25:
thank you for you reply , the below is what I have revised: protobuf caused the checksum failure
```
diff --git a/WORKSPACE b/WORKSPACE
index b8c257d..cbdc038 100644
--- a/WORKSPACE
+++ b/WORKSPACE
@@ -338,8 +338,8 @@
 github_archive(
     name = "protobuf",
     repository = "google/protobuf",
-    commit = "v3.1.0",
-    sha256 = "fb2a314f4be897491bb2446697be693d489af645cb0e165a85e7e64e07eb134d",  # noqa
+    commit = "v3.2.0",
+    sha256 = "a839d3f1519ff9d68ab908de5a0f269650ef1fc501c10f6eefd4cae51d29b86f",  # noqa
 )
diff --git a/tools/workspace/bitbucket.bzl b/tools/workspace/bitbucket.bzl
index 7691b5b..ae052af 100644
--- a/tools/workspace/bitbucket.bzl
+++ b/tools/workspace/bitbucket.bzl
@@ -43,8 +43,8 @@ def bitbucket_archive(
     # bucket.
     mirrors = [
         "https://bitbucket.org/%s/get/%s.tar.gz",
-        "https://drake-mirror.csail.mit.edu/bitbucket/%s/%s.tar.gz",
-        "https://s3.amazonaws.com/drake-mirror/bitbucket/%s/%s.tar.gz",
+        #"https://drake-mirror.csail.mit.edu/bitbucket/%s/%s.tar.gz",
+        #"https://s3.amazonaws.com/drake-mirror/bitbucket/%s/%s.tar.gz",
     ]
     urls = [mirror % (repository, commit) for mirror in mirrors]
 
diff --git a/tools/workspace/github.bzl b/tools/workspace/github.bzl
index bbe8403..6a8aeb0 100644
--- a/tools/workspace/github.bzl
+++ b/tools/workspace/github.bzl
@@ -41,8 +41,8 @@ def github_archive(
     # Packages are mirrored from GitHub to CloudFront backed by an S3 bucket.
     mirrors = [
         "https://github.com/%s/archive/%s.tar.gz",
-        "https://drake-mirror.csail.mit.edu/github/%s/%s.tar.gz",
-        "https://s3.amazonaws.com/drake-mirror/github/%s/%s.tar.gz",
+    #    "https://drake-mirror.csail.mit.edu/github/%s/%s.tar.gz",
+    #    "https://s3.amazonaws.com/drake-mirror/github/%s/%s.tar.gz",
     ]
     urls = [mirror % (repository, commit) for mirror in mirrors]
 

```

And I have got these results:
```
bob-ros@bobros-Think:~/drake/drake-distro$ bazel build //... 
ERROR: build interrupted
INFO: Elapsed time: 28.644s
FAILED: Build did NOT complete successfully (0 packages loaded)
    currently loading: tools/vector_gen ... (5 packages)
bob-ros@bobros-Think:~/drake/drake-distro$ bazel test //...  
INFO: Analysed 4565 targets (101 packages loaded).
INFO: Found 1850 targets and 2715 test targets...
INFO: From Executing genrule //drake/automotive/models:speed_bump_genrule:
[2017-11-25 13:43:31.462] [console] [info] Loading road geometry.
[2017-11-25 13:43:31.463] [console] [info] Generating OBJ.
FAIL: //tools/workspace:bazel_codestyle (see /home/bob-ros/.cache/bazel/_bazel_bob-ros/aeb2cce4621bd070fcf8d1a0ba4ce510/execroot/drake/bazel-out/clang-4.0-linux-opt/testlogs/tools/workspace/bazel_codestyle/test.log)
INFO: From Testing //tools/workspace:bazel_codestyle:
==================== Test output for //tools/workspace:bazel_codestyle:
tools/workspace/github.bzl:44:5: E122 continuation line missing indentation or outdented
tools/workspace/github.bzl:45:5: E122 continuation line missing indentation or outdented
================================================================================
INFO: Elapsed time: 4678.326s, Critical Path: 220.92s
INFO: Build completed, 1 test FAILED, 14763 total actions
//tools/workspace:bazel_codestyle                                        FAILED in 0.2s
  /home/bob-ros/.cache/bazel/_bazel_bob-ros/aeb2cce4621bd070fcf8d1a0ba4ce510/execroot/drake/bazel-out/clang-4.0-linux-opt/testlogs/tools/workspace/bazel_codestyle/test.log

Executed 2715 out of 2715 tests: 2714 tests pass and 1 fails locally.
There were tests whose specified size is too big. Use the --test_verbose_timeout_warnings command line option to see which ones these are.
bob-ros@bobros-Think:~/drake/drake-distro$ bazel build --compiler=gcc-5 //... 
INFO: Analysed 4565 targets (1 packages loaded).
INFO: Found 4565 targets...
INFO: From Executing genrule //drake/automotive/models:speed_bump_genrule:
[2017-11-25 16:13:23.451] [console] [info] Loading road geometry.
[2017-11-25 16:13:23.452] [console] [info] Generating OBJ.
INFO: Elapsed time: 4199.711s, Critical Path: 124.29s
INFO: Build completed successfully, 12013 total actions
```

jamiesnape, 2017-11-26:
Looks like just a lint error with the edits you needed to make to `github.bzl`, so basically everything is good now, it seems.

jwnimmer-tri, 2017-12-15:
Seems resolved; please re-open if there are still problems.

closed 2017-12-15 by jwnimmer-tri