RobotLocomotion/drake #7518: Weird construction of VectorXd in solvers/create_constraint.cc

joaospinto, 2017-11-23:
I'm using part of your library (within a separate build system), and get a number of compiler errors on solvers/create_constraint.cc, essentially due to initializations of Eigen::VectorXd objects being done similarly to `Eigen::VectorXd x{n};` instead of `Eigen::VectorXd x(n);` (where `n` is an int). The compiler seems to think that we are trying to construct a vector of size 1 with unique element `y.size()`. Typically the error looks like this:
> error: narrowing conversion of ‘n’ from ‘const long unsigned int’ to ‘double’ inside { } [-Werror=narrowing]
I would imagine this is not resulting in wrong behavior, but thought I should note it here. Probably what's happening is that you compile it using a less pedantic compiler mode and the code ends up doing the right thing.

jamiesnape, 2017-11-23:
And you are compiling with `-std=c++14` (or something similar)?

joaospinto, 2017-11-23:
Yes. I guess this works fine with C++11?

jamiesnape, 2017-11-23:
No, we require C++14, so it should be fine with `-std=c++14`. Running a Bazel build on Mac with `-Werror=narrowing` added did not trigger any errors for me.

joaospinto, 2017-11-27:
This essentially amounts to the following simple example:
```
#include <Eigen/Dense>
#include <iostream>
int main() {
    Eigen::VectorXd v{2};
    std::cout << v << std::endl;
    std::cout << v.size() << std::endl; 
}
```
I tried to build this with G++ and Clang, with -std=c++{11,14}, and it always worked fine (printed "0 0 2" instead of "0 1"). I would have imagined this would print "0 1", but clearly I was wrong. Essentially the reason is that `Eigen::VectorXd` does *not* have list initialization (however, when the size is determined at compile-time, e.g. 'Eigen::Matrix<double, 2, 1>', it does have list initialization!), so using the curly braces will just call the constructor that takes the size of the vector as a single argument.
I don't see what is to be gained in using the curly braces here, but functionally it's doing the right thing.
I still don't understand why I got the error in my case, but I think this is not relevant for other people, so will close this issue.

closed 2017-11-27 by joaospinto